UECIDE/UECIDE #60: Formats special characters to garbage characters

avluis, 2015-01-09:
v0.8.7z28 on OS X 10.2.2
I have only being able to reproduce this with this character, but it doesn't seem to like the degrees (°) symbol at the moment.
Whenever it encounters that file it will progressively change it until it becomes garbage;
Ex:
````c
// 13 is 100k Hisens 3950 1% up to 300°C for hotend "Simple ONE" & "Hotend "All In ONE" (original line)
// 13 is 100k Hisens 3950 1% up to 300�C for hotend "Simple ONE" & "Hotend "All In ONE" (after closing and reopening same file)
````

majenkotech, 2015-01-09:
On OSX it's not saving the file as UTF-8 like it does on Linux.  The special characters, which are represented as UTF-8, don't get saved right.

I just write the raw content of the edit area to file, so this might be an issue with the RSyntaxTextArea widget.

majenkotech, 2015-01-09:
I have forced UTF-8 file format when writing the data to file.  It seems to have fixed it.

avluis, 2015-01-09:
Thanks! That was quick!

CONSULitAS, 2015-02-01:
Same issue on Windows, too.

If you copy content from GitHub for Windows to UECIDE editor everything looks fine. If you save, it will become garbage, because not saved as UTF-8. :-(

@majenkotech How can I force UTF-8 when writing? Did not find it. Thanks.

majenkotech, 2015-02-01:
What version are you running?  If you're still on j then you can expect it to fail, yes, because you don't have the fix.  Make sure you're running the latest beta. This is an old bug now.

CONSULitAS, 2015-02-09:
@majenkotech THANKS!  It does work now like expected.

It has been the V j of course. On the download page the beta is a little bit hard to find. :-)

majenkotech, 2015-02-09:
I have added a special note to the top of the download page now saying there are known issues with the stable version and to try the beta.

closed 2015-02-09 by majenkotech

CONSULitAS, 2015-02-09:
looks good! :+1: