UECIDE/UECIDE #61: How to select different programmers.

pitstopz, 2015-02-08:
I know this must be totally simple but I can't seem to figure out where to add or select additional programmers. I only have 2 options upload to Arduino Direct or via USBasp. I have a USBtiny but can not figure out how to add it as select-able item from within uecide.
Btw thanks for the hard work on this. I love this level of IDE vs ATmel Studio Eclipse, etc.

CHeers
Dave

majenkotech, 2015-02-08:
There's two things you'd need to add: the programmer specification, and an entry in the available programmers list.  This can either be done in the board.txt for the board or the core.txt file for the core.

If you look in the core.txt file you can see the existing programmer specifications, in the form:

    programmer.<key>.name=<Pretty Name>
    programmer.<key>.using=<serial / hid>
    programmer.<key>.command.<os>=command::and::arguments::for::upload
    ... etc ...

I suggest copying one of the existing ones and adjusting as necessary, changing the key of course.

Then you need to add the key to the `sketch.upload` key, either in the core.txt or in the board.txt if you only want it available to one (or a few) boards.

Also, if you want to share the command used to do the uploading I can add it as a programmer in the main repository too.

pitstopz, 2015-02-10:
Thanks for the reply. Was never able to find the section:

programmer.<key>.name=<Pretty Name>
programmer.<key>.using=<serial / hid>
programmer.<key>.command.<os>=command::and::arguments::for::upload
... etc ...

in any of the core .txt files.

I was able to find this and modified as below:

upload.usbtiny.name=Upload using USBTiny
upload.usbtiny.command.linux_arm=${core.root}/tools/linux_pi/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux_pi/avrdude.conf::-cstk500v2::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.linux_amd64=${core.root}/tools/linux64/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux64/avrdude.conf::-cstk500v2::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.linux_i386=${core.root}/tools/linux32/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux32/avrdude.conf::-cstk500v2::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.macosx=${core.root}/tools/macosx/avrdude::-p${build.mcu}::-Pusb::-b1::-C${core.root}/tools/macosx/avrdude.conf::-cstk500v2::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.windows=${core.root}/tools/windows/avrdude::-p${build.mcu}::-Pusb::-b1::-C${core.root}/tools/windows/avrdude.conf::-cstk500v2::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.verbose=-v::-v::-v
upload.usbtiny.quiet=-q::-q
upload.protocol=arduino

sketch.upload=arduino::usbasp::usbtiny

This adds an entry to select USBTiny as a programmer from GUI but yields
the following error:

avrdude: usbdev_open(): did not find any USB device "usb"
Upload Failed

When I run  the avrdude and conf file included with UECIDE from the
terminal command line it works fine.

Thanks
Dave

majenkotech, 2015-02-10:
Sorry that should have been upload.

I can't understand why it would work from the terminal but not from the IDE. Find files and programs that are provided in the core it should work the same in both. Which operating system are you using ?

pitstopz, 2015-02-10:
Linux Mint 17

majenkotech, 2015-02-10:
When you're running from the terminal are you doing it as root, or through sudo?

pitstopz, 2015-02-10:
standard user

pitstopz, 2015-02-10:
Same user profile as UECIDE runs from.

majenkotech, 2015-02-10:
Looking at examples online (on Adafruit) the command line doesn't use the `-P usb` directive, nor `-c stk500v2`. Instead it uses `-c usbtiny`.  I don't know enough about the usbtiny to know even how it's meant to work ;)  Try changing it to use that instead and see if it makes a difference.

pitstopz, 2015-02-10:
The -P needs to remain usb but the -c did indeed need to be usbtiny.
It is working perfectly now. I sent paypal 15 pounds as a thank you.
I am very active in our local MakerSpace and have been encouraging some
users to try out
UECIDE. Best of luck with the app.
Cheers,
Dave

majenkotech, 2015-02-10:
Thank you very much :)  Your support is greatly appreciated.    If you can provide the final working additions for the usbtiny I can incorporate them in the next release of the core so everyone gets them.

pitstopz, 2015-02-10:
Sure. Attached is my Arduino core.txt.

name=arduino
description=Arduino 1.0.5
compiler=avr-gcc

core.path=api
core.flags=-DF_CPU=${build.f_cpu}::-DARDUINO=${core.version}::-D${board}
core.header=Arduino.h
core.version=105
version=1.0.5
revision=9

core.library.api=api

family=avr

library.path=libraries
examples.path=examples

board=_UNKNOWN_BOARD_

upload.arduino.name=Upload direct to an Arduino board
upload.arduino.command.freebsd=${core.root}/tools/freebsd/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/freebsd/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.command.linux_arm=${core.root}/tools/linux_pi/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/linux_pi/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.command.linux_amd64=${core.root}/tools/linux64/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/linux64/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.command.linux_i386=${core.root}/tools/linux32/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/linux32/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.command.macosx=${core.root}/tools/macosx/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/macosx/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.command.windows=${core.root}/tools/windows/avrdude::${if:${board.erase}=yes,,-D}::-V::-p${build.mcu}::-P${port}::-b${upload.speed}::-C${core.root}/tools/windows/avrdude.conf::-c${upload.protocol}::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.arduino.using=serial
upload.arduino.verbose=-v
upload.arduino.quiet=-q::-q
upload.arduino.dtr=yes
upload.arduino.rts=yes

upload.usbasp.name=Upload using USBasp
upload.usbasp.command.freebsd=${core.root}/tools/freebsd/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/freebsd/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.command.linux_arm=${core.root}/tools/linux_pi/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/linux_pi/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.command.linux_amd64=${core.root}/tools/linux64/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/linux64/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.command.linux_i386=${core.root}/tools/linux32/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/linux32/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.command.macosx=${core.root}/tools/macosx/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/macosx/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.command.windows=${core.root}/tools/windows/avrdude::-B10::-p${build.mcu}::-C${core.root}/tools/windows/avrdude.conf::-cusbasp::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbasp.using=hid
upload.usbasp.verbose=-v
upload.usbasp.quiet=-q::-q

upload.usbtiny.name=Upload using USBTiny
upload.usbtiny.command.linux_arm=${core.root}/tools/linux_pi/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux_pi/avrdude.conf::-cusbtiny::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.linux_amd64=${core.root}/tools/linux64/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux64/avrdude.conf::-cusbtiny::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.linux_i386=${core.root}/tools/linux32/avrdude::-p${build.mcu}::-Pusb::-B1::-C${core.root}/tools/linux32/avrdude.conf::-cusbtiny::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.macosx=${core.root}/tools/macosx/avrdude::-p${build.mcu}::-Pusb::-b1::-C${core.root}/tools/macosx/avrdude.conf::-cusbtiny::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.command.windows=${core.root}/tools/windows/avrdude::-p${build.mcu}::-Pusb::-b1::-C${core.root}/tools/windows/avrdude.conf::-cusbtiny::-Uflash:w:${build.path}/${filename}.hex:i::${verbose}
upload.usbtiny.verbose=-v::v::-v
upload.usbtiny.quiet=-q::-q
upload.protocol=arduino

sketch.upload=arduino::usbasp::usbtiny

icon.128=icons/arduino128.png
icon.64=icons/arduino64.png
icon.48=icons/arduino48.png
icon.32=icons/arduino32.png
icon.22=icons/arduino22.png
icon.16=icons/arduino16.png

manual=reference

majenkotech, 2015-02-10:
Added.  Cheers.

closed 2015-02-10 by majenkotech