UECIDE/UECIDE #62: Total crash of the Plugin manager with 0.8.8alpha1 (OSX)

madias123, 2015-02-16:
As posted into the uecide forum:
sadly on OSX 10.9 the plugin manager crashes (just clicking on any entry in the tree) without the possibility to repair. To fix it, the only possibility is to delete all files in the User/xxx/Libraries and go back to an old version and reinstall everything again. So I would recommend not to update to this version yet.

Java log file:
OS: macosx_i386
Version: 0.8.8alpha1
Build: 565
Exception: 
java.lang.NoClassDefFoundError: org/markdown4j/Markdown4jProcessor
	at org.uecide.plugin.PluginManager$11.valueChanged(PluginManager.java:383)

deladriere, 2015-02-16:
I reverted to Version 0.8.7z36 but have still the same problem

madias123, 2015-02-16:
If you can't wait for a fix, to this, but you'll ve to reinstall everything:
delete uecide from the program folder
delete the whole order: User/<your name>/Libraries/UECIDE
reinstall the older version
update all plugins.

majenkotech, 2015-02-16:
Just booting my OS X VM to experiment to see what's happening.

majenkotech, 2015-02-16:
From other comments it looks like it may be the markdown processor that is not being found.

majenkotech, 2015-02-16:
Ok, so it was just a packaging problem.  I forgot to add the markdown jar file to github so it never got packaged with the distribution build.  New build running now.

majenkotech, 2015-02-16:
New version is now up. Works on my OSX VM.

madias123, 2015-02-16:
Thank, you rebuilding should work now.
But there are some boards missing?
For example:
Whole TI stuff
Arduino Due
maple (please don't up it anymore, it's broken :) )
See Pic at:
http://666kb.com/i/cw6imszmyz7ogr7e3.gif

majenkotech, 2015-02-16:
Have you read the message at the top of the beta page?  It explains that it's using a new repo system and not all the packages are converted yet.

madias123, 2015-02-16:
Do humans ever mind big/fat/bold signs? ;)

Seriously: Please do not add the leaflabs/maple stuff again. It's too old, and it won't work on a modern system (didn't got it to work on any OS) . We are developing the new STM32-Arduino API (running on Arduino 1.6), so this should be the better candidate for the whole STM32Fxxx systems.

majenkotech, 2015-02-16:
That sounds great.  The maple port was low priority anyway, so I'll look to use the newer 1.6 core instead.  Would it be this repo? https://github.com/madias123/Arduino_STM32

madias123, 2015-02-16:
Better bookmark this (my fork isn't up to date and more experimental)
https://github.com/rogerclarkmelbourne/Arduino_STM32
I will drop you a line, when we'll be ready!
Other thing:
On chipkit the whole low level stuff isn't found/linked: Some errors:
error: 'PPS_OUT_REFCLKO' was not declared in this scope
error: 'mapPps' was not declared in this scope
error: 'PPS_OUT_SDO1' was not declared in this scope

majenkotech, 2015-02-16:
Forked.

Which board is that error for?

madias123, 2015-02-16:
DP32 or chipkit UNO, but maybe I shoot my whole UECIDE installation. (But I think not)

majenkotech, 2015-02-16:
Shouldn't give messages like that for UNO - that has no PPS in it.  I'll check the DP32 defs.

madias123, 2015-02-16:
These were only examples, like error: 'REFOCONbits' was not declared in this scope, so the whole stuff not working. more examples:
variable or field '__ISR' declared void
error: 'INT_ENABLED' was not declared
'INTSetVectorPriority' was not declared

majenkotech, 2015-02-16:
Sounds like you have some crossover incompatibility there.  You should really start from a clean system with this new repo.

deladriere, 2015-02-16:
But they are also gone from Version 0.8.7z36 ?
Can't see the Calunium anymore for example
Maybe because I installed the Alpha before reinstalling Version 0.8.7z36 
?

majenkotech, 2015-02-16:
The plugin manager is what defines what repo you look at.  You have the newer "alpha" plugin manager installed, so the old version of the IDE uses that.  If you want to run the two side by side you will need to implement two separate data directories.

deladriere, 2015-02-16:
Thx !
Back on track by deleting the plugin manager

madias123, 2015-02-16:
nop. no luck at all with wiping all: These are all the errors:
Compiling...
• Compiling sketch...
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino: In function 'void setup()':
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:92:22: error: 'INT_SPI_1_VECTOR' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:92:40: error: 'INT_PRIORITY_LEVEL_4' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:92:60: error: 'INTSetVectorPriority' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:93:43: error: 'INT_SUB_PRIORITY_LEVEL_0' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:93:67: error: 'INTSetVectorSubPriority' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:94:11: error: 'INT_SPI1TX' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:94:23: error: 'INT_ENABLED' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:94:34: error: 'INTEnable' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino: At global scope:
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:103:11: error: variable or field '__ISR' declared void
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:103:27: error: 'ipl4' was not declared in this scope
/Users/madias/Documents/uecide/PIC32/I2s/CS4334/CS4334-begin3/CS4334-begin3.ino:147:1: error: expected '}' at end of input
Failed compiling sketch

majenkotech, 2015-02-16:
Good-o.  When this new plugin manager / repo system is properly off the ground it's going to revolutionize things somewhat.

Right, your code is using code for the old compiler. We're now running by default on the newer compiler.  There are some differences and some symbols not defined.

You don't use __ISR any more - you use just:

    void __attribute__((interrupt)) myIsr() {...}

And just use numbers for the priorities.

deladriere, 2015-02-16:
can't wait to use your library repo
will it organize itself for the appropriate boards ?

majenkotech, 2015-02-16:
Kind of, yes.  If you change the drop-down at the top it filters the libraries to just show the ones for that family.

madias123, 2015-02-16:
hm, I must rethink all my IRQ's....
I2C looks also broken: (tested on uno/DP32)
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp: In constructor 'DTWI0::DTWI0()':
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp:1429:105: error: '_DTWI0_SCL_PIN' was not declared in this scope
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp:1429:121: error: '_DTWI0_SDA_PIN' was not declared in this scope
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp: In constructor 'DTWI1::DTWI1()':
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp:1454:105: error: '_DTWI1_SCL_PIN' was not declared in this scope
/Users/madias/Library/uecide/libraries/Communications/chipkit/Wire/DTWI.cpp:1454:121: error: '_DTWI1_SDA_PIN' was not declared in this scope
Failed compiling libraries

madias123, 2015-02-16:
and why is there a "plugin manager" under plugins?

majenkotech, 2015-02-16:
Because the plugin manager itself is a plugin - though I am going to be moving it back into the core instead so I can do more magical things.

It looks like some of the board definitions in Ricklon's 1.6 core may be out of date - just updating.

madias123, 2015-02-16:
Is there a documentation for setting up interrupts with the new compiler? Some things are clear to me, but not all....
so my old:
INTSetVectorPriority(INT_SPI_1_VECTOR, INT_PRIORITY_LEVEL_4);
	INTSetVectorSubPriority(INT_SPI_1_VECTOR, INT_SUB_PRIORITY_LEVEL_0);
	INTEnable(INT_SPI1TX, INT_ENABLED);
...
void __ISR(_SPI_1_VECTOR, ipl4) SPI1InterruptHandler(void)

will get to:
   setIntVector(_SPI_1_VECTOR, (isrFunc)SPI1InterruptHandler);
   setIntPriority(_SPI_1_VECTOR, 4, 0);
setIntEnable(INT_SPI1TX);
....
void __attribute__((interrupt(),????)) SPI1InterruptHandler(void) {

majenkotech, 2015-02-16:
Pretty much.  The function though is just:

    void __attribute__((interrupt)) SPI1InterruptHandler() {
        //code in here
    }

No need to specify the priority or anything in the interrupt, just in the setIntPriority call. 

There is a document being drafted at the moment - it's going through the proof reading stage at the moment I believe.

majenkotech, 2015-02-16:
Oh, and the symbols you want are:

    _SPI_1_VECTOR

and

    _SPI1_TX_IRQ

madias123, 2015-02-16:
ok, thank you, I think, I've got it:
setIntVector(_SPI_1_VECTOR, (isrFunc)SPI1InterruptHandler);
setIntPriority(_SPI_1_VECTOR, 4, 0);
setIntEnable(_SPI1_TX_IRQ);
}
....
void __attribute__((interrupt)) SPI1InterruptHandler() { // I2s Data out
               SPI1BUF =bit24; // something to write
 clearIntFlag(_SPI1_TX_IRQ); 
	}
Compiles without error, maybe it will function also :)

one last question:
setIntPriority(_SPI_1_VECTOR, 4, 0); means: vector priority 4, sub priority 0 - right?

thank you
Matthias

majenkotech, 2015-02-16:
Yep, you got it :)

It's also good practice to put a `clearIntFlag()` just before the `setIntEnable()` call to ensure the flag is cleared and doesn't trigger immediately you enable the interrupt.

madias123, 2015-02-16:
yippie! now i'll check this with my oszi. We definitely need a good documentation for this. The real power of the PIC32's are the whole SPI/I2s/... stuff, I'd like to setup nearly everything by hand (Hey, microchip! I need a PIC32 with 8 SPI ports!) :)

majenkotech, 2015-02-16:
MZ has 6... ;)  Not far off...

madias123, 2015-02-16:
the MZ has everything...but no floating point, if I remember correctly. That's horrible! Ok, maybe with PIC32MZ EF...

majenkotech, 2015-02-16:
Yeah, that is one drawback of it.  It would be nice to have hardware FPU.  Still, with the EBI I guess you could interface an external one...

majenkotech, 2015-02-16:
Maybe dig out an ancient MC68882 ;)  Only 50MHz though...

madias123, 2015-02-16:
I think of an "one (main)chip virtual synth with analogue filters/VCA's": 
with 6x SPI: 
3x SPI(I2s) = 6 independent Voices (audio DAC's)
3x SPI with octal DAC's   = 24 channel of (8 or 10bit) control voltage for using it with "real" VCA and VCF's
and now with a FPU you'll get the sweetest and craziest sound out of it.
a cheap TFT and some encoders+buttons: ready is the next kickstarter project for an unbelievable price. You must think, the cheapest 4-voice hybrid-digital-analog synth, the Dave smith tetra costs about 700 Euros!

majenkotech, 2015-02-16:
You could use 6 SPI for 12 voices on DACs, and then use the EBI to interface to as many parallel ADCs as you like.

closed 2018-10-26 by majenkotech