UECIDE/UECIDE #67: New board def for pinguino micro

sker65, 2015-02-19:
I've done another board definition for the olimex pinguino micro board.

How can I contribute?

br, Steve

majenkotech, 2015-02-19:
At the moment I am in a transition phase between repository systems.  Best method at the moment is to attach the files zipped up to a post on the forum. I can then take them and turn them into the right packages for distribution.

sker65, 2015-02-20:
Hi Matt,

I've done a posting in the forum but the forum system says: "you post
need approval" and I can't attach anything?

Could you please check this.

Thanks Steve

majenkotech, 2015-02-20:
I have authorized your user - you should be able to attach now.

sker65, 2015-02-20:
Thanks, I've added the board zip file.

br, steve

closed 2018-10-26 by majenkotech