UECIDE/UECIDE #71: Problems with a PIC32 USB STARTER KIT 2

Dagui34, 2015-03-25:
Hi everybody, so i have a PIC32 USB STARTER KIT 2. In the plugin manager of UECIDE i downloaded the requirement.
But when i plug my Pic32 board it is recognised by windows (7 pro 64) as a "Custom USB Device" and it doesn't appears in UECIDE as an available board.

An Idea ?

Best regards, Dagui34.

majenkotech, 2015-03-25:
The Microchip boards are a direct import from MPIDE.  I don't know how most of them operate.  It looks like this one has a custom debugger circuit on it (PIC32MX440 based) which will only work with Microchip's closed source MPLAB-X. I am not aware of any command line programming software for that interface.

The only way you will be able to use it with UECIDE is to install a chipKIT bootloader (USB version) custom compiled for that board and use the USB OTG port to connect it to your computer.  Another option would be to connect a picKIT2 to the ICSP header and program using that, which is supported by UECIDE.

There has actually been talk of removing all the Microchip boards from MPIDE because they can't be directly programmed without jumping through hoops first.

Dagui34, 2015-04-09:
Hello Majenkotech, thanks for your answer.

Effectively it would be a good idea to remove the products when it's not compatible.

Best regards, David.

majenkotech, 2015-04-09:
The thing is, there are ways of making it compatible - just not easy ones and they require reprogramming or otherwise specially preparing the board.    It's just a shame Microchip don't like open sourcing their programmers so the community could then write support software for them.

Another option may be (if it's supported by the built-in programmer) would be to replace the firmware on it with the "Microchip Scripting Host" firmware which them makes it act like a pickit2. That way it could be programmed directly with pic32prog in UECIDE.

majenkotech, 2015-04-13:
You might find this of interest: http://chipkit.net/programming-microchip-starter-kits-mpide/

Dagui34, 2015-04-13:
Hello Majenkotech !

This is very useful for many people.
Thanks a lot to spend time.

Personally I will sell my 2 PIC32 USB Starter Kit and their devboard to concentrate myself on my Picadillo35T board. I love it.

Best regards, David.

closed 2018-10-26 by majenkotech