UECIDE/UECIDE #77: how to print

queenidog, 2015-11-07:


queenidog, 2015-11-07:
How do I print hardcopies of my programs?  There is no print dialog in UEDecide.  Otherwise, still pretty good IDE, still evaluating.  But I need hardcopy.

majenkotech, 2015-11-07:
There is no print functionality. You are only the second person in the history of UECIDE to ask after it, so it is a very very low priority.

If you need to print then just copy and paste the text into something like Word and set it to a fixed width font, like Courier.

closed 2015-11-07 by majenkotech

queenidog, 2015-11-07:
Hmm, your program is the only one in history not to have a print function!

 

How about a debug program to run in the MCU, so we can see registers, outputs, and single-step through the program?  Has anyone asked for this?  (My MCSP does this, reduces troubleshooting time 90%).  Using println is so yesterday.

 

Copy and paste doesn’t keep the formatting and it looks like hell and unreadable.  I tried Wordpad and Word.

 

I did copy it to MicroCode Studio Plus where the formatting is retained but the colors aren’t quite the same.  Not a solution.

 

I can do a screen print but that’s awkward.

 

If I can’t print I can’t use the program.  Simple as that.  

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

majenkotech, 2015-11-07:
Register-level debugging has been asked for, and it simply cannot be done. If UECIDE were an IDE dedicated to a specific family of chips and understood what those chips were and how they worked then it may be possible. But, as it's a generic IDE that doesn't know one chip from another it is impossible.

Printing will come one day, but it's such a pain to implement it in Java that I just don't have time. If you have any Java skills to spare then feel free to contribute.