UECIDE/UECIDE #78: How to use a proxy?

rolfn, 2015-11-25:
I try the newest (beta) version but the program doesn't start because
```
Error downloading http://repo.uecide.org/uecide/dists/uecide/cores/
binary-linux-amd64/Packages.gz
```
I am behind a proxy server which is needed for all internet connections. It seems to me that UECIDE doesn't read and doesn't use my environment variables ` http_proxy`/`https_proxy`. What can I do?

majenkotech, 2015-11-25:
You need to tell Java about your proxy, not just Linux.  It's stupid, I know, but that's Java for you...

For example: 

    export _JAVA_OPTIONS='-Dhttp.proxyHost=my.proxy.host.com -Dhttp.proxyPort=3128'

rolfn, 2015-11-25:
Thank you for your quick response. I tried your suggestion. I added the line
```
export _JAVA_OPTIONS='-Dhttp.proxyHost=webproxy -Dhttp.proxyPort=8080'
```
to the start script but it's still not working:
```
Picked up _JAVA_OPTIONS: -Dhttp.proxyHost=webproxy -Dhttp.proxyPort=8080
org.reflections.ReflectionsException: could not get type for name org.uecide.plugin.SerialTerminal
        at org.reflections.ReflectionUtils.forName(ReflectionUtils.java:378)
        at org.reflections.ReflectionUtils.forNames(ReflectionUtils.java:387)
        at org.reflections.Reflections.getSubTypesOf(Reflections.java:338)
        at org.uecide.Base.loadPlugins(Base.java:1890)
        at org.uecide.Base.<init>(Base.java:473)
        at org.uecide.Base.main(Base.java:161)
Loading icon sets...
Icon file org/uecide/icons/winston.icon = Winston
Icon file org/uecide/icons/gnomic.icon = Gnomic
Icon file org/uecide/icons/mono.icon = Monochrome
Tree background: java.awt.Color[r=255,g=255,b=255]
No boards installed: You have no boards installed.  I will
now open the plugin manager so you can
install the boards, cores and compilers
you need to use UECIDE.
```
The original Arduino IDE works without additional definition (see "Preferences" / "Network" / "Auto-detect proxy settings")

rolfn, 2015-11-25:
I tried it with the stable version with no success, then again with the beta version. Now it work's. Thank you!

majenkotech, 2015-11-25:
There is code that I can add to the IDE itself to manage proxies from within it, but I think you are one of about 10 people left in the world that still use antiquated proxy systems ;)

Plus I don't know how yet ;)

closed 2015-11-25 by rolfn