UECIDE/UECIDE #81: External editor command syntax?

TCWORLD, 2016-01-16:
Hi,

I see you can just right click on a file and select "open in external editor" and in the preferences there is the option for "External Editor command". 

What is the syntax of this command?

For example if I want to open something in Notepad++, I presume the command would start with something like:

    "C:\Program Files (x86)\Notepad++\notepad++.exe"

But if the command is just this, all that happens is Notepad++ opens, but doesn't open the file I want. For NP++, the file to open should be passed as an argument to the above command. Is there a wildcard that should be placed in the UECIDE preferences that indicates the filename?

majenkotech, 2016-01-16:
The command is passed through the same string parser as everything else in UECIDE, and the tag `${filename}` is set to the filename to edit.  It is in the same format as the contents of all the configuration and control files with `::` to separate arguments so you don't need quotes. You get access to pretty much every setting in UECIDE through those tags - everything in the core, board, compiler, etc.

So the command would be:

    C:\Program Files (x86)\Notepad++\notepad++.exe::${filename}

TCWORLD, 2016-01-16:
Great, thanks. Works perfectly.

closed 2016-01-16 by TCWORLD

deladriere, 2016-01-17:
on Mac with Atom it works too with 
/Applications/Atom.app/Contents/MacOS/Atom::${filename}