UECIDE/UECIDE #82: UECIDE Unable to find jar file (Windows)

TCWORLD, 2016-02-02:
Hi,

Having upgraded to UECIDE 0.8.7z36, I've noticed a strange issue. It wasn't present in 0.8.7j.

Basically when I run the UECIDE exe file directly everything works as expected. However, if I create a shortcut and try to run it, things don't work so well, I get an error message saying "unable to open jar file lib\uecide.jar".

If I change the shortcut to have the "Start In" setting as the path to the UECIDE.exe file (e.g. <path>\<to>\<uecide>\), then the shortcut works. However if I pin it to the start menu, it shows me a list of recent files (windows thing), but trying to open them directly from there doesn't respect the "Start In" path for the shortcut.

Did something change between the two versions in how you generated the exe?

majenkotech, 2016-02-02:
Yes, I certainly did change the executable at some point, and it may have been around that time (I can't recall off hand, I'd have to check the commit logs). 

I moved from using an ancient version of launch4j which didn't allow the passing of command line parameters to Java to using my own home-written wrapper program written in Visual C++. 

This is certainly not something I have ever come across, but then I have never "pinned" it to the start menu (I'm not even sure what that means).

Have you tried the beta version 0.8.8alpha22? I most likely made some changes in the executable between the versions which may have had an affect on it.

Also in the beta version there is a "bundled" version which is a single exe file that contains everything within it - including the jar file. The one thing it doesn't contain is Java itself, so you need to ensure you have Java installed on your computer for it to run.  That bundled file should just run directly from wherever it is in whatever way you run it.  That one is built using a much more modern version of launch4j specifically for its bundling facilities. TBH though I can't recall if it allows the use of command line parameters in that version :/

TCWORLD, 2016-02-02:
I'll try the latest beta version and see if that helps. Will report back when I've had a chance to try it.

majenkotech, 2016-02-02:
I have actually rejigged the website today. There is no stable or beta, only the current version.

closed 2018-10-26 by majenkotech