UECIDE/UECIDE #83: plugins no longer work in older versions

jazz1138, 2016-02-05:
Hi,

Can you please add the www alias to your website it prevents plugins from working in older versions of website. And google takes you to a non-existant web page.

Jarrad

majenkotech, 2016-02-05:
The www alias is already there. What isn't are the old plugins.

I am no longer supporting the old version of UECIDE. It is too old and the plugins were all too out of date.

Please uograde to the current version.

jazz1138, 2016-02-08:
Thanks, We are using this in a Windows student lab environment. I will try to upgrade to the latest version once I  have resolved some of the problems with our devices and issues with folder redirection under Windows. For the moment we will just get students to use the old version and download the plugins manually.

closed 2018-10-26 by majenkotech