UECIDE/UECIDE #85: windows 10

jlbachiochi, 2016-03-02:
Using UECIDE with Picadillo-35T (MAX32)

[image]

While there is no complaint about 'not finding' Picadillo.h,it can't seem to find the Picadillo class.  Wondering if it has somethng to do with where windows 10 is expecting to find things.

majenkotech, 2016-03-02:
You have the wrong board selected. The Picadillo code only gets compiled if you have the picadillo board selected. That is because it contains register references that aren't common to all PIC32 chips (and certainly not AVR ATMega2560 which you have selected at the moment). Just select the Picadillo as your board and it should compile fine.

jlbachiochi, 2016-03-02:
Yes, you are correct I selected both Picadillo-35t and Chipkit.  Unselecting the Chipkit button fixed my issue.  I look forward to working with UECIDE  --jeff

From: Majenko Technologies 
Sent: Wednesday, March 2, 2016 9:06 AM
To: UECIDE/UECIDE 
Cc: jlbachiochi 
Subject: Re: [UECIDE] windows 10 (#85)

You have the wrong board selected. The Picadillo code only gets compiled if you have the picadillo board selected. That is because it contains register references that aren't common to all PIC32 chips (and certainly not AVR ATMega2560 which you have selected at the moment). Just select the Picadillo as your board and it should compile fine.

closed 2018-10-26 by majenkotech