UECIDE/UECIDE #86: Crash on board setting load

Bogusmips, 2016-03-26:
Hello,

I encounter a NPE with the last commit (771290f522045515610e216d082497a8ea95ac76) under Debian Jessie, I can see that the programmer object has replaced a string in the sketch class.

The crash occur when a board is selected, if none (i.e. "board=" in the preferences file) the IDE is starting normally (the programmer line is the same as before "board.arduino-mega2560.programmer=arduino").

Error message:
Sorry, I cannot select a valid programmer.
java.lang.NullPointerException
        at org.uecide.Sketch.setProgrammer(Sketch.java:395)
        at org.uecide.Sketch.setProgrammer(Sketch.java:371)
        at org.uecide.Sketch.setCompiler(Sketch.java:362)
        at org.uecide.Sketch.setCore(Sketch.java:328)
        at org.uecide.Sketch.setBoard(Sketch.java:284)
        at org.uecide.Sketch.setBoard(Sketch.java:253)
        at org.uecide.Sketch.loadSketchFromFolder(Sketch.java:568)
        at org.uecide.Sketch.<init>(Sketch.java:225)
        at org.uecide.Sketch.<init>(Sketch.java:164)
        at org.uecide.Base.createNewEditor(Base.java:1294)
        at org.uecide.Base.handleNew(Base.java:1243)
        at org.uecide.Base.<init>(Base.java:763)
        at org.uecide.Base.main(Base.java:165)

majenkotech, 2016-03-26:
I think this might be something that I fixed the other day (but not checked in). Try my latest commit, see if that fixes it for you.

Bogusmips, 2016-03-26:
Indeed, seems to be fixed :)
Thanks.

closed 2016-03-26 by Bogusmips

majenkotech, 2016-03-26:
Incidentally, I have just uploaded the programmers that I have so far converted to the new object based system to the master repository, so you should be able to install them now.