UECIDE/UECIDE #87: can't upload launchpad g2553

dave2x11, 2016-05-13:
error output from UCEIDE

Unable to start process
java.io.IOException: Cannot run program "C:\Users\david\AppData\Local\UECIDE\cores\energia-msp430/tools/windows/mspdebug" (in directory "C:\Users\david\AppData\Local\Temp\build-65145d02-bade-4f3d-9e53-31966ff15b83"): CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(Unknown Source)
	at org.uecide.Context.runSystemCommand(Context.java:867)
	at org.uecide.Context.executeCommand(Context.java:542)
	at org.uecide.Context.executeKey(Context.java:485)
	at org.uecide.Sketch.programFile(Sketch.java:3470)
	at org.uecide.Sketch.upload(Sketch.java:1650)
	at org.uecide.Editor$DefaultRunHandler.run(Editor.java:184)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(Unknown Source)
	at java.lang.ProcessImpl.start(Unknown Source)
	... 8 more
• Resetting board.
Upload Failed

majenkotech, 2016-05-15:
I have updated the core. It should fix this problem.

dave2x11, 2016-05-16:
I downloaded UCEIDE 0.8.8 again and got the same error message except for a different built number.

part of the error message:   Cannot run program "C:\Users\david\AppData\Local\UECIDE\cores\energia-msp430/tools/windows/mspdebug"

/tools/windows/mspdebug seems to be missing from the energia-msp430 directory

David

majenkotech, 2016-05-16:
Did you update the core?

dave2x11, 2016-05-16:
I am a novice in this. I thought that by downloading UCEIDE again I would have the updated core. If that is not true, how do I update the core:

David

majenkotech, 2016-05-16:
I the plugin manager where you installed it in the first place.

dave2x11, 2016-05-16:
After going to the plugin manager, it took me awhile to realize that I had to click on refresh before I could go on.

The results are better but not totally fixed. Uploading starts but does not finish.

the error message:

Uploading firmware...
• Resetting board.
• Uploading...
MSPDebug version 0.20 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2012 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

usbutil: unable to find a device matching 0451:f432

• Resetting board.
Upload Failed

majenkotech, 2016-05-16:
And what is the VID and PID of your board?

dave2x11, 2016-05-16:
Where do I look for that data?

majenkotech, 2016-05-16:
In the device manager of your chosen operating system

dave2x11, 2016-05-16:
Is this what you want?

USB\VID_0451&PID_F432&REV_0105&MI_00
USB\VID_0451&PID_F432&MI_00

David

majenkotech, 2016-05-16:
That looks correct. I do not know why mspdebug isn't finding it. Which operating system are you using?

dave2x11, 2016-05-17:
Windows 7

dave2x11, 2016-05-17:
Also, I just used Energia with the same program and everything was fine.

majenkotech, 2016-05-17:
Can you check the version of mspdebug in Energia and find the precise command that it is running with all its parameters?

dave2x11, 2016-05-17:
compiler and mspdebug output from energia:

C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-g++ -c -Os -Wall -ffunction-sections -fdata-sections -mmcu=msp430g2553 -DF_CPU=16000000L -MMD -DARDUINO=101 -DENERGIA=17 -IC:\energia-0101E0017\energia-0101E0017\hardware\msp430\cores\msp430 -IC:\energia-0101E0017\energia-0101E0017\hardware\msp430\variants\launchpad C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\sketch_may16a.cpp -o C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\sketch_may16a.cpp.o
[C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-g++, -c, -Os, -Wall, -ffunction-sections, -fdata-sections, -mmcu=msp430g2553, -DF_CPU=16000000L, -MMD, -DARDUINO=101, -DENERGIA=17, -IC:\energia-0101E0017\energia-0101E0017\hardware\msp430\cores\msp430, -IC:\energia-0101E0017\energia-0101E0017\hardware\msp430\variants\launchpad, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\sketch_may16a.cpp, -o, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\sketch_may16a.cpp.o]
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\atof.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\dtostrf.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\itoa.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\random.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\twi.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\twi_sw.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\usci_isr_handler.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\WInterrupts.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\wiring.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\wiring_analog.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\wiring_digital.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\wiring_pulse.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\wiring_shift.c.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\HardwareSerial.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\IPAddress.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\MACAddress.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\main.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\new.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\Print.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\Stream.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\TimerSerial.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\Tone.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\Wire.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\WMath.cpp.o
  Using previously compiled: C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\WString.cpp.o
C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar rcs C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\atof.c.o
[C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar, rcs, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\atof.c.o]
C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar rcs C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\dtostrf.c.o
[C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar, rcs, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\dtostrf.c.o]
C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar rcs C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\itoa.c.o
[C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar, rcs, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\itoa.c.o]
C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar rcs C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\random.c.o
[C:\energia-0101E0017\energia-0101E0017\hardware\tools\msp430\bin\msp430-ar, rcs, C:\Users\david\AppData\Local\Temp\build1419842458462271026.tmp\core.a, C:\Users\david\AppData\Local\Temp\build1419842458462271026. [... cut ...]
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

MSP430_GetNumberOfUsbIfs
MSP430_GetNameOfUsbIf
Found FET: HID0011:COM9
MSP430_Initialize: HID0011:COM9
Firmware version is 20409001
MSP430_VCC: 3000 mV
MSP430_OpenDevice
MSP430_GetFoundDevice
Device: MSP430G2xx3 (id = 0x00de)
2 breakpoints available
MSP430_EEM_Init
Chip ID data: 25 53
Erasing...
Programming...
Writing 2526 bytes at c000...
Writing   32 bytes at ffe0...
Done, 2558 bytes total
MSP430_Run
MSP430_Close

majenkotech, 2016-05-17:
OK, I have upgraded mspdebug to the same version as in energia and made a couple of configuration tweaks. Do a full upgrade of all your plugins to see if it now works.

dave2x11, 2016-05-17:
Good job. It worked.

Now I will try some other Launchpads.

David

dave2x11, 2016-05-17:
I tried another Launchpad FR5969. At first it did not load. Then I tried Energia and had to update the board firmware.

After that both Energia and UCEIDE worked.

David

majenkotech, 2016-05-17:
Interesting...  I have an FR6989 that I have never managed to get going. I just updated the programmer and now it works...

I guess I should be adding the ability to do that into UECIDE somehow. It's simple enough though - `mspdebug tilib --allow-fw-update` - just need to work out the best way of including it in UECIDE for msp430 only.