UECIDE/UECIDE #90: Can't register for forim access

vanepp, 2016-07-21:
I tried to register as vanepp in the forums but the confirmation email never arrived. I just tried to reregister and it tells me the user name and email address are both in use. I clicked on the resend 
confirmation email button on the register page but still no confirmation email. I can't see a way in the
forums page to contact the administrator as it suggests so I'm trying here.

majenkotech, 2016-07-21:
Your account has been activated. U do not know where the activation email has gone. It may have been diverted as spam somewhere along the line.

vanepp, 2016-07-21:
Thanks, first time I've seen an activation email get lost and 
unfortunatly I'm retired and no longer have access to the sendmail logs to 
see if it was SFU's system that lost it.

Peter Van Epp

majenkotech, 2016-07-21:
The fault is at my end - looking into it now.

majenkotech, 2016-07-21:
Got it! That was fun. The email is relayed via Google's Gmail servers. Which I had all set up properly and working. Then suddenly it started connecting to the servers through an IPv6 address instead of the IPv4 address I had configured, so it was being rejected as coming from an unauthorized source.

So now it works. 

Still, it's helped keep the spammers at bay...

closed 2016-07-21 by majenkotech