UECIDE/UECIDE #95: Version 0.8.9-pre4 Can't load blink to Arduino Uno

vanepp, 2016-08-04:
Version 0.8.9-pre4 Win7 pro Arduino Uno on com3 installed and configured (including the ardude programmer which wasn't picked up by just installing the Uno board). Trying to compile the blink
example (which works on pre3) gets this on the console:

• Uploading sketch... 
avrdude: can't open input file C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/Blink.hex: No such file or directory
avrdude: read from file 'C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/Blink.hex' failed

--- debug console ---

Sketch.java 306 (setCore): Selecting core Arduino 1.0.x
Sketch.java 348 (setCompiler): Selecting compiler GCC 4.8.1 for AVR
Sketch.java 662 (createBuildFolder): Creating build folder build-45b28512-a85e-4ef1-899f-321968ffbc16
PropertyFile.java 281 (save): Saved property file c:\Users\Owner\AppData\Local\UECIDE_pre\preferences.txt
PropertyFile.java 281 (save): Saved property file c:\Users\Owner\AppData\Local\UECIDE_pre\preferences.txt
PropertyFile.java 281 (save): Saved property file c:\Users\Owner\AppData\Local\UECIDE_pre\preferences.txt
Editor.java 2309 (heading): Compiling...
Editor.java 2323 (bullet): Preprocessing...
Editor.java 2323 (bullet): Compiling sketch...
Context.java 937 (runSystemCommand): Execute: c:\Users\Owner\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-g++ -c C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16\Blink.cpp -o C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16\Blink.cpp.o -mmcu=atmega328p -g -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -DF_CPU=16000000L -DARDUINO=106 -DARDUINO_UNO -Ic:\Users\Owner\AppData\Local\UECIDE_pre\cores\arduino\arduino -Ic:\Users\Owner\AppData\Local\UECIDE_pre\boards\Arduino\arduino-uno -IC:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16 -Ic:\Users\Owner\AppData\Local\UECIDE_pre\cores\arduino\examples\01.Basics\Blink -Os -w 
Editor.java 2323 (bullet): Compiling core...
Editor.java 2337 (bullet2): arduino
Editor.java 2323 (bullet): Compiling libraries...
Editor.java 2323 (bullet): Linking sketch...
Context.java 937 (runSystemCommand): Execute: c:\Users\Owner\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-gcc -Wl,--gc-sections -Os -w -mmcu=atmega328p -o C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/Blink.elf C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16\Blink.cpp.o -LC:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16 -Lc:\Users\Owner\AppData\Local\UECIDE_pre\cache\arduino\arduino-uno -Wl,--start-group -lCore_arduino -Wl,--end-group -lm 
Editor.java 2309 (heading): Compiling done.
Editor.java 2309 (heading): Memory usage
Context.java 937 (runSystemCommand): Execute: c:\Users\Owner\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-size C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/Blink.elf 
Editor.java 2323 (bullet): Program size: 1030 bytes
Editor.java 2323 (bullet): Memory size: 9 bytes
Editor.java 2323 (bullet): Compilation took 0.232 seconds
Editor.java 2309 (heading): Uploading firmware...
Editor.java 2323 (bullet): Resetting board... 
Editor.java 2323 (bullet): Uploading sketch... 
Context.java 937 (runSystemCommand): Execute: c:\Users\Owner\AppData\Local\UECIDE_pre\programmers\avrdude-arduino/windows/avrdude -D -V -patmega328p -PCOM3 -b115200 -Cc:\Users\Owner\AppData\Local\UECIDE_pre\programmers\avrdude-arduino/windows/avrdude.conf -carduino -Uflash:w:C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/Blink.hex:i -q -q 
PropertyFile.java 281 (save): Saved property file c:\Users\Owner\AppData\Local\UECIDE_pre\preferences.txt
PropertyFile.java 281 (save): Saved property file c:\Users\Owner\AppData\Local\UECIDE_pre\preferences.txt

--- listing of 

C:\Users\Owner\AppData\Local\Temp\build-45b28512-a85e-4ef1-899f-321968ffbc16/

Blink.cpp
Blink.cpp.o
Blink.elf

no .hex file. So I expect there is a conversion step missing in the compile chain somewhere.

Luckily I haven't yet upgraded one of my other machines from pre3 yet (and probably won't for now!)

Version 0.8.9-pre3

 debug.log ( no uno attached to the machine which will have been what the excepction is about.

Context.java 937 (runSystemCommand): Execute: c:\Users\Peter\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-gcc -Wl,--gc-sections -Os -w -mmcu=atmega328p -o C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.elf C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa\Blink.cpp.o -LC:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa -Lc:\Users\Peter\AppData\Local\UECIDE_pre\cache\arduino\arduino-uno -Wl,--start-group -lCore_arduino -Wl,--end-group -lm 
Context.java 937 (runSystemCommand): Execute: c:\Users\Peter\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.elf C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.eep 
Context.java 937 (runSystemCommand): Execute: c:\Users\Peter\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-objcopy -O ihex -R .eeprom C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.elf C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.hex 
Editor.java 2299 (heading): Compiling done.
Editor.java 2299 (heading): Memory usage
Context.java 937 (runSystemCommand): Execute: c:\Users\Peter\AppData\Local\UECIDE_pre\compilers\avr-gcc/bin/avr-size C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.elf 
Editor.java 2313 (bullet): Program size: 1030 bytes
Editor.java 2313 (bullet): Memory size: 9 bytes
Editor.java 2313 (bullet): Compilation took 3.355 seconds
Editor.java 2299 (heading): Uploading firmware...
Editor.java 2313 (bullet): Resetting board... 
Base.java 2440 (error): 
Base.java 2441 (error): ******************** EXCEPTION ********************
Base.java 2442 (error): An uncaught exception occurred:
Base.java 2443 (error):     The cause is: null
Base.java 2444 (error):     The message is: null
Base.java 2445 (error): 
Base.java 2449 (error):         org.uecide.builtin.port.main(port.java:52)
Base.java 2449 (error):         sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
Base.java 2449 (error):         sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
Base.java 2449 (error):         sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
Base.java 2449 (error):         java.lang.reflect.Method.invoke(Unknown Source)
Base.java 2449 (error):         org.uecide.Context$1.run(Context.java:864)
Base.java 2453 (error): ******************** EXCEPTION ********************
Base.java 2454 (error): 

listing of 
C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/Blink.elf C:\Users\Peter\AppData\Local\Temp\build-ba4e6150-a81f-4fab-86bd-b085f85db7fa/

Blink.cpp
Blink.cpp.o
Blink.eep
Blink.hex

---

console:

Compiling...
• Preprocessing...
• Compiling sketch...
• Compiling core...
? arduino
• Compiling libraries...
• Linking sketch...
Compiling done.
Memory usage
• Program size: 1030 bytes
• Memory size: 9 bytes
• Compilation took 3.355 seconds
Uploading firmware...
• Resetting board... 

it didn't appear to notice the exception on the UECIDE console, it just sits there at resetting the board
until I close it.

majenkotech, 2016-08-04:
Ah - I need to push a new version of the compiler. I changed how the compilation works to make it more flexible in pre4. It requires a couple of extra directives in the compiler config file to tell it what files to generate from the .elf file. The .hex is of course one of those.  Building the packages now - it'll be a bit before they have uploaded.

In the meantime you can add these lines to compilers/avr-gcc/compiler.txt:
````
compile.autogen.eep=${objcopy}::-O::ihex::-j::.eeprom::--set-section-flags=.eeprom=alloc,load::--no-change-warnings::--change-section-lma::.eeprom=0::${build.path}/${filename}.elf::${build.path}/${filename}.eep
compile.autogen.hex=${objcopy}::-O::ihex::-R::.eeprom::${build.path}/${filename}.elf::${build.path}/${filename}.hex
compile.autogen.lss=__builtin_stdin::${build.path}/${filename}.lss::${objdump}::-h::-S::${build.path}/${filename}.elf
compile.autogen.etx=__builtin_stdin::${build.path}/${filename}.etx::${objdump}::-t::${build.path}/${filename}.elf
````

It doesn't matter where in the file you put them.

majenkotech, 2016-08-05:
OK, new compiler packages uploaded. Updating them in the plugin manager should fix it.

vanepp, 2016-08-05:
Yep, refresh then update all got new compiler and it now loads blink just fine. Fast work! Thanks!