UECIDE/UECIDE #96: Scaling with High DPI displays

maaaatti, 2016-08-23:
Windows 10.
3200x1800.
UECIDE seems to report itself to Windows as High DPI aware, and thus the interface appears tiny. Unusable.

majenkotech, 2016-08-23:
This is actually a known issue of Java, not a problem with UECIDE.

You can try experimenting with some of the fixes shown here:

http://superuser.com/questions/988379/how-do-i-run-java-apps-upscaled-on-a-high-dpi-display

majenkotech, 2018-10-26:
All editor fonts are now editable, and certain theme options have font size settings.

closed 2018-10-26 by majenkotech

majenkotech, 2018-11-25:
You should use the current 0.10.0-RC1 and check out the theming options in preferences - especially the **JTattoo Font Size** option (Theme/JTattoo Options) when using JTattoo-based themes.  You can set a Giant font which makes everything in the interface bigger. You can also change all the editor and console fonts to something bigger.

High DPI is a real bugbear for Java applications. There's really nothing I can do about it, since I have no control over how Swing renders its widgets without reimplementing them all from scratch, which I'm not about to do.