UECIDE/UECIDE #97: Suggestion: Ubuntu PPA

sbrl, 2016-10-16:
Having a PPA for Ubuntu would make it really easy for Linux users to stay up to date.

majenkotech, 2016-10-16:
I did start looking at having a PPA but the build system (ant) doesn't seem to work too well with launchpad. (https://launchpad.net/uecide)

Instead you just need to manually add my repository to /etc/apt/sources.list

    deb http://majenko.co.uk/debian stable main
    deb-src http://majenko.co.uk/debian stable main

For testing versions:

    deb http://majenko.co.uk/debian unstable main
    deb-src http://majenko.co.uk/debian unstable main

closed 2016-10-16 by majenkotech

sbrl, 2016-10-16:
You have a repository? That'll do just fine! What's the package name? The name `uecide` doesn't show up for me if I run an `apt search uecide`.

majenkotech, 2016-10-16:
Actually thinking about it only the "unstable" version is there.

majenkotech, 2016-10-16:
There, the "alpha22" version should be in "stable" now.

sbrl, 2016-10-16:
Yes! Found it! Thanks so much. It's just so much easier to use a repo / ppa to install stuff. Perhaps the website should show this under the linux instructions?

Also, off topic but how do you package uecide as a .deb and host your own repo? I can't seem to find any decent instructions on the process anywhere.

sbrl, 2016-10-16:
Erm oh dear. I get an impossible situation if I try to install that package:

```
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 uecide : Depends: openjdk-7-jdk but it is not installable
E: Unable to correct problems, you have held broken packages.
```

majenkotech, 2016-10-16:
It is a bit of a black art. You need to create a "debian" directory and fill it with the right files. You can create a template structure for a project with "dh_make", do it manually, or copy and modify files from an existing project.

Then you can use `dpkg-buildpackage`.  It's quite an involved process getting it right.

Everything in UECIDE is deb based - all the plugins, etc, are .deb files.

majenkotech, 2016-10-16:
I thought I'd fixed that in that deb. It should be using default-jdk. Hold on...

majenkotech, 2016-10-16:
Ok, it might work now... You may need to do some purging of caches to pick up the change since the version is the same as before.

sbrl, 2016-10-17:
Oooh thanks! It does work now :smile: