UECIDE/UECIDE #98: Request: Support for the Wemos D1 R2

sbrl, 2016-10-17:
Now that I've managed to install this lovely program, I've been poking around the plugin manager. I can't find the `Wemos D1 R2`. Is it supported? Where can I find it?

Thanks :smile:

majenkotech, 2016-10-17:
I'm a bit out of date with the ESP8266 boards. I need to spend some time
doing an update and making sure it all compiles and works, and add all the
new boards that people have made.

sbrl, 2016-10-17:
Ah cool - I'll wait for the update then :smile: 

For some background, the Wemos D1 R2 is an ESP8266 board based on the ESP8266EX. It appears to have several cousins, such as the Wemos D1 (without the `R2`) and the D1 mini Pro, although I don't know the difference between the former and the D1 R2.

GeoffState, 2017-09-05:
I have installed today and also need Wemos D1 Mini Pro support. I didn't find it in the plugins either and I notice the Majenko-Technologies eBay store no longer exists. Maybe the UECIDE project has died due to lack of support.

majenkotech, 2017-09-05:
Far from died, my friend, far from it. The shop closed because eBay increased their fees and it became too expensive to have a shop for the few items I had on there (eBay shops aren't free). The items for sale are still there, just no shop. The shop is now on majenko.co.uk

The only reason I don't have D1 support yet is I don't have a D1 to work with. That and I am only one man with limited time.

Of course since this is all open source there's nothing stopping someone else crafting and contributing the board files...

majenkotech, 2017-09-05:
I now have the D1 Mini, D1 Mini Lite and D1 Mini Pro on order from Ali Express. It'll take a month to crawl its way across the globe to me.

GeoffState, 2017-09-05:
Wicked, you shop where I shop. What does a board file look like, maybe I could craft a D1 Mini Pro for you.

majenkotech, 2017-09-05:
They are basically an esp8266 on a carrier. It's a question of which one is
on what and how the pins are mapped...

sbrl, 2017-09-06:
Oooh the pins are _nasty_. They _aren't_ as they are described any of the diagrams I've found (if you find one do let me know"), _or_ as they are described on the board itself - at least in the arduino IDE (though that might be because I chose "D1 (old)" instead of "D1 R2"..... :P

closed 2018-11-12 by majenkotech