UECIDE/UECIDE #102: Windows 10 UEDIDE won't run.

grtbluyonder, 2017-01-04:
uecide-0.8.8alpha22-full.exe,  Trying to load program, I get error message: **could not open C:\program files\(x86)\java\jre 1.8.0_60\Lib\jvm.cfg**   and app fails to load.

Doesn't appear in program files anywhere so didn't install properly.  Reinstall does not solve issue.

majenkotech, 2017-01-04:
It sounds like your java installation is broken. I suggest you reinstall it, or uninstall it so that UECIDE uses the builtin Java installation as a fallback.

grtbluyonder, 2017-01-05:
Thanks, I did an uninstall then re-install with same failure. I'll try just
an uninstall.

Serveurperso, 2017-10-30:
The ZIP all in one package run perfectly tested on two different windows 10 computers.
I created a "portable" directory to get an all in one portable app (new feature) :D

closed 2018-10-26 by majenkotech