UECIDE/UECIDE #103: Linker script not found

Bamaan, 2017-04-24:
I'm trying to compile for a Fubarino Mini, but can't resolve this error:

`sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
pic32-cpp: error: ERR: No such file or directory
pic32-cpp: warning: '-x c' after last input file has no effect
pic32-cpp: fatal error: no input files
compilation terminated.
pic32-cpp: error: ERR: No such file or directory
pic32-cpp: warning: '-x c' after last input file has no effect
pic32-cpp: fatal error: no input files
compilation terminated.
/home/user/.uecide/compilers/pic32-tools/bin/bin/gcc/pic32mx/4.8.3/../../../../bin/pic32-ld: cannot open linker script file ERR.00: No such file or directory
collect2: error: ld returned 255 exit status
Compiling Failed`

I removed uecide and reinstalled it with manually deleting the .uecide folder.
I use chipkit-core and PIC32 compiler v1.43.

From the Arduino IDE the project compiles. (But then fails if I try to compile again, but that's a different  problem) So I might be able to use the compiler from the Arduino folder.
Does anyone have an idea of where this error could originate?

Thanks

majenkotech, 2017-04-24:
This is actually down to the version of Java you have. Java have replaced the old javascript engine (Rhino) with a new one (Nashorn) in Java 8. In order for old javascript code to work on the new engine you have to `load("nashorn:mozilla_compat.js");` - however that isn't liked by Rhino (there is no "load" function).

I thought I'd worked out a workaround for it, but obviously it got lost somewhere in the mix (or it never actually worked, I forget which now).

So there's two things you can try:

1. Upgrade your java installation to java 8.
2. Delete the contents of "usr/share/uecide/system/vc/" in the UECIDE data directory.

The second option will cause UECIDE to fall back on hard-coded Java versions of the "VC" commands. At the moment there should be no difference between the two versions of the commands, but I won't be adding any new commands as I come up with them to the Java core, only the javascript vc folder.

I shall re-investigate how to make it not do the "load" function call on Rhino, if it is even possible.

majenkotech, 2017-04-24:
Ok, it looks like that was actually easier than I was expecting. Simply wrapping load() in a try{}catch{} block does the trick by all accounts.

A new version of the core-varcmd package has been uploaded and can be updated in the plugin manager - that should make it work properly on older versions of Java now.

Bamaan, 2017-04-24:
It works! Thank you

closed 2017-04-24 by Bamaan