UECIDE/UECIDE #105: PIC32 broken on MacOSX v0.8.9-pre10

DanielO, 2017-05-10:
Hi,
I downloaded v0.8.9-pre10 for OSX to compile a PIC32 project but it doesn't work. The same code compiles (and works :) on Windows.

This can be reproduced by using the chipKIT Max32 board and trying to compile the servo 'Knob' example.

```
Output is as follows..
Compiling...
• Preprocessing...
• Compiling sketch...
• Compiling core...
‣ api
• Compiling libraries...
‣ Servo
• Linking sketch...
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (find#32) in find at line number 32
pic32-cpp: error: ERR: No such file or directory
pic32-cpp: warning: '-x c' after last input file has no effect
pic32-cpp: fatal error: no input files
compilation terminated.
pic32-cpp: error: ERR: No such file or directory
pic32-cpp: warning: '-x c' after last input file has no effect
pic32-cpp: fatal error: no input files
compilation terminated.
/Users/oconnd1/Library/uecide/compilers/pic32-tools/bin/bin/gcc/pic32mx/4.8.3/../../../../bin/pic32-ld: cannot open linker script file ERR.00: No such file or directory
collect2: error: ld returned 255 exit status
Compiling Failed
```

majenkotech, 2017-05-10:
This is already addressed in an update to the system-varcmd package. You just need to update your plugins. If you can't get the GUI running you can do it from the command line:

$ java -jar /Applications/UECIDE.app/Contents/Resources/Java/uecide.jar --update --upgrade

DanielO, 2017-05-12:
I ran that and got this...
```
[w7au06130517c5l 10:57] ~/projects/Fh@tBadge >java -jar /Applications/UECIDE.app/Contents/Resources/Java/uecide.jar --update --upgrade
Error downloading /org/uecide/dist/dists/uecide/uecide/binary-darwin-amd64/Packages.gz: null
Upgrading repos-master...Fetching http://repo.uecide.org/uecide/pool/repos/r/repos-core/repos-master_1.0.1_all.deb
...80%...100%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%Extracting repos-master_1.0.1_all.deb
0%...done
Upgrading core-varcmd...Fetching http://repo.uecide.org/uecide/pool/extra/c/core-varcmd/core-varcmd_1.0.4_all.deb
...10%...30%...40%...60%...70%...80%...90%...100%0%...10%...20%...30%...40%...50%...60%...70%...80%...90%...100%Extracting core-varcmd_1.0.4_all.deb
0%...10%...20%...30%...40%...50%...60%...70%...done
```

However it had no effect on the compilation, i.e. it failed as above.