UECIDE/UECIDE #106: MacOSX 0.8.9-pre11 is broken

DanielO, 2017-05-10:
When I tried and run this app OSX says it's damaged and refuses to run it.

If I run the java stub binary manually it works so maybe it's an easy fix..

```
[Maelstrom 9:44] ~ >/Volumes/UECIDE/UECIDE.app/Contents/MacOS/JavaApplicationStub
[JavaAppLauncher] Requested [1.5*], launching in [1.6] instead.
javax.script.ScriptException: sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
	at com.sun.script.javascript.RhinoScriptEngine.invoke(RhinoScriptEngine.java:230)
	at com.sun.script.javascript.RhinoScriptEngine.invokeFunction(RhinoScriptEngine.java:188)
	at org.uecide.Context.executeJavaScript(Context.java:640)
	at org.uecide.Context.runFunctionVariable(Context.java:840)
	at org.uecide.Context.parseString(Context.java:800)
	at org.uecide.Context.executeKey(Context.java:564)
	at org.uecide.Context.executeUScript(Context.java:731)
	at org.uecide.Context.executeKey(Context.java:586)
	at org.uecide.Base.runInitScripts(Base.java:3268)
	at org.uecide.Base.<init>(Base.java:735)
	at org.uecide.Base.main(Base.java:170)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:39)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:25)
	at java.lang.reflect.Method.invoke(Method.java:597)
	at apple.launcher.LaunchRunner.run(LaunchRunner.java:116)
	at apple.launcher.LaunchRunner.callMain(LaunchRunner.java:51)
	at apple.launcher.JavaApplicationLauncher.launch(JavaApplicationLauncher.java:52)
javax.script.ScriptException: sun.org.mozilla.javascript.internal.EcmaError: ReferenceError: "load" is not defined. (prefs#33) in prefs at line number 33
	at com.sun.script.javascript.RhinoScriptEngine.invoke(RhinoScriptEngine.java:230)
	at com.sun.script.javascript.RhinoScriptEngine.invokeFunction(RhinoScriptEngine.java:188)
	at org.uecide.Context.executeJavaScript(Context.java:640)
	at org.uecide.Context.runFunctionVariable(Context.java:840)
	at org.uecide.Context.parseString(Context.java:800)
	at org.uecide.Context.executeKey(Context.java:591)
	at org.uecide.Context.executeUScript(Context.java:731)
	at org.uecide.Context.executeKey(Context.java:586)
	at org.uecide.Base.runInitScripts(Base.java:3268)
	at org.uecide.Base.<init>(Base.java:735)
	at org.uecide.Base.main(Base.java:170)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:39)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:25)
	at java.lang.reflect.Method.invoke(Method.java:597)
	at apple.launcher.LaunchRunner.run(LaunchRunner.java:116)
	at apple.launcher.LaunchRunner.callMain(LaunchRunner.java:51)
	at apple.launcher.JavaApplicationLauncher.launch(JavaApplicationLauncher.java:52)
Unable to find /Volumes/UECIDE/ERR
Showing changelog as there is a new version
Service 'Tree Updater' started
Service 'Network Discovery' started
Service 'USB Device Discovery' started
Service 'Changed Files' started
```

majenkotech, 2017-05-10:
This is already addressed in an update to the system-varcmd package. You just need to update your plugins. If you can't get the GUI running you can do it from the command line:

$ java -jar /Applications/UECIDE.app/Contents/Resources/Java/uecide.jar --update --upgrade

majenkotech, 2017-05-10:
Actually, before you do that, I could do with some testing - I don't have a mac to test on unfortunately.

Is it possible for you to edit the file `/Applications/UECIDE.app/Contents/Resources/Info.plist` and find the line:

    <key>JVMVersion</key>
    <string>1.5*</string>

And change it to:

    <key>JVMVersion</key>
    <string>1.8*</string>

If that works I can get it to request a newer version of Java which will cure so many problems with OS X.

DanielO, 2017-05-12:
Oddly even after I edit that file it still says..
```
[w7au06130517c5l 10:59] ~/projects/Fh@tBadge >/Volumes/UECIDE/UECIDE.app/Contents/MacOS/JavaApplicationStub
[JavaAppLauncher] Requested [1.5*], launching in [1.6] instead.
Showing changelog as there is a new version
2017-05-12 11:00:17.945 JavaApplicationStub[20455:1923029] -_cgsEventRecord is deprecated for _cgsEventRecord. Please use -CGEvent.
```

Even deleting that stanza has no effect, very odd :(

majenkotech, 2017-05-12:
Hmmm, that is very odd. I tried it on my Slow Leopard VM and it requested 1.8* and was offered 1.6 (the latest available on that ancient system).

I have a couple of other fixes that want pushing out, so I am going to bundle a new version with this change in today - you might try that and see what it makes of it...

majenkotech, 2017-07-16:
I have created a new version of UECIDE (-pre15) which has replaced the (rather aged) stock JavaApplicationStub with a script file that works much better. For a start it finds the Oracle Java installations, not just the Apple ones, so should be far more reliable.  It can also be enhanced to allow the bundling of my own version of Java inside the application (as I do in the Windows "full" version) should it become necessary in the future.

DanielO, 2017-07-17:
This appears to work much better, thanks.

I can start and compile my sketch with no problems :)

I don't have the hardware to hand right now but it certainly seems to be fixed.

closed 2017-07-17 by DanielO