UECIDE/UECIDE #107: chipkit WF32 not compiling basic sketch

samchik, 2017-05-29:
all other chipkit uno32 WifiFire and max all compiling basic sketch 
no compiling on chipkit WF32 window nor Mac please help also if include sofswerial will not see it but it does on any other board .
on the beta Alpha or the old version working pretty good !
thank you

majenkotech, 2017-05-30:
This is the fault of the linker script. For some bizarre reason Digilent decided not to include the correct linker script in the chipKIT core, but keep it separate, even though it's a completely standard (identical to the '795) linker script. When I built my own copy of the core I used to copy that file in, but now I build direct from the Ardino packages it's not there.

majenkotech, 2017-05-30:
I have added the files to the core and made a PR. Once they get pulled the chipkit-nightly package should contain them within a day.

* https://github.com/chipKIT32/chipKIT-core/pull/354

samchik, 2017-07-03:
Thank you for the help 
It compile basic sketch but I'm facing unkown issue that my sketch compile okay no errors but won't run like the way it runs on my other computer ! 
Both computer 32bit and identical I even got another third identical to try to copy the files from the running computer and still no lock , so now have three computer two of them on the new uecide version program run and download to chipkitwf32 or chipkit wifire but program won't run correctly 
And one older version runs perfect 
uecide pic32 compiler for mx and mz version 4.8.3 
Beta  I have no problem what so ever except crashing one a while but I'm okay with it , I don't know why this new version have me running around and chasing my tail from problem to another. I wish you have all versions history available for download. 
Is there any way to get this exact version because my program so long and big takes forever to change it for the new version and what not !

samchik, 2017-07-05:
Still trying to figure out why working code and same hardware not working any more on any of the Pcs that I install the new uecide lates version 
Only one pc with UECIDE alpha beta version , chipKIT core, compiler for MX and MZ 4.8.3,is working 
I even copy that version and install on a different pc for back up doesn't work I don't even have that compiler max and mz 
I don't understand why 
In side the program the string serial work for 
If (strcmp(buff,"FF04"==o)
But will not work in 
If (StringOne.startwith("FF04",0)) 
I don't get it work on the UECIDE 0.8.alpha beta version chipKIT core, compiler for MX and MZ 4.8.3,
But not on any of the new version. 
The chipkit boards 
Chipkit wf32 and wifire boards 
Is there any clue or ideas I tried every thing I could

majenkotech, 2017-07-05:
It sounds like you are using really old versions of everything there. You should be on 0.9.8pre14 with chipKIT core 1.4.1 and compiler "PIC32 Compiler version 1.43"

If something worked on one specific computer but not on others it generally means there is some bug in your code that just happened to compile in such a way that it worked that time. Trying to compile on another computer or with another version of software may well cause it to compile differently and your bug shows itself as your program not working right.

I cannot support all the old versions of cores and things, though I do try my best to do what I can. The current repos have all the compilers right back to the C30 based GCC 4.5.2, but whether that works with the current core (which is very much out of my control) I can't say.

closed 2018-10-26 by majenkotech