UECIDE/UECIDE #108: Arch Linux Package?

sbrl, 2017-08-05:
I've just installed Manjaro (and Arch Linux distribution), and I've discovered that there isn't an arch linux package for UECIDE. Would an AUR package be possible?

If nobody else is able to help here, I'll tackle this myself once I've gotten myself more acquainted with Manjaro.

majenkotech, 2017-08-05:
I have never had anything to do with arch or manjaro. I wouldn't know where to begin. Maybe alien can convert the Deb file, though it's not a tool I have used for many years, even decades.

majenkotech, 2017-08-05:
No, it looks like alien only handles deb, rpm and slp (whatever that is). No mention of aur packages.

sbrl, 2017-08-05:
@majenkotech No worries then! I'll gain some experience myself and then look at puuting something together myself :smiley_cat:

majenkotech, 2017-08-05:
I spent the morning installing Arch and finally, after 3 attempts, got it to boot right. Horrible system. Nasty. Manjaro looks like it may be easier, but when building packages I like to go to the source for best support.

So I managed to cobble together a PKGBUILD script and made a package. No idea what it does ;)

It's on the releases page along with the PKGBUILD script if you want to try it out.

* https://github.com/UECIDE/UECIDE/releases/tag/0.8.9-pre16

sbrl, 2017-08-05:
@majenkotech Oh, wow! Congrats. I didn't even anwant to go _near_ vanilla arch after the stories I've heard.

I'll have to get back to you on what a PKGBUILD file does and how to use it, as I've not much more of a clue than you have :P

jonathonf, 2018-02-27:
There are AUR packages now:

* https://aur.archlinux.org/packages/uecide/
* https://aur.archlinux.org/packages/uecide-git/