UECIDE/UECIDE #109: Serial Terminal Not Working

kdahlhaus, 2018-08-17:
I am able to compile and upload a sketch to the Arduino Nano.    It writes to Serial.  I can view this output in the normal Arduino IDE on the same port.  If I close the Arduino IDE, and openUECIDE, the Serial Terminal always prints "** Port Closed **" and no output appears.  I can close UECIDE and open the Arduino IDE to the same COM port and see the output.

I've looked through the UECIDE and UECIDE_plugins source and do not see that text.

Running Version 0.9.5 of the IDE on Windows 10.

Any idea what is happening here?

majenkotech, 2018-08-17:
I have seen someone else recently showing the same issue. I was unsure if it was actually the code on the board giving the problem. This means it may be the serial terminal then instead. That's good to know. I will dig into it and see what it is. It's a Windows-only problem, of course...

majenkotech, 2018-08-17:
BTW, the message is emitted from: https://github.com/UECIDE-Plugins/serialterminal/blob/master/src/org/uecide/plugin/SerialTerminal.java#L658

However, that is just a function called from [here](https://github.com/UECIDE-Plugins/serialterminal/blob/master/src/org/uecide/plugin/SerialTerminal.java#L124).

It seems `port.exists()` isn't ever becoming true. Not sure why. The `exists` function just looks for the existence of a file (https://github.com/UECIDE/UECIDE/blob/master/src/org/uecide/SerialCommunicationPort.java#L610). My guess is that looking for the existence of "COM3" doesn't work like every other operating system in the world does. Maybe something else is needed for Windows esoterica.

majenkotech, 2018-08-17:
God I hate windows...

majenkotech, 2018-08-17:
I have disabled it in testing on Windows - exists always returns true. However, disconnect the board with the serial terminal open and the port never works again until you restart UECIDE.

majenkotech, 2018-08-17:
I can't think any more tonight. I'll sleep on it and see what I can come up with over the weekend.

kdahlhaus, 2018-08-18:
OK thanks for looking at it.