UECIDE/UECIDE #110: Standard window functionality support

sensboston, 2018-10-06:
Could you please add standard Windows functionality to the UECIDE main window? Now program for some strange reasons modified standard window menu, and only "Close window" choice is available (no "Pin to taskbar", no grouping etc.)

I don't know why this strange functionality was implemented, but it's kinda unuseful and annoying.

majenkotech, 2018-10-06:
I don't know what "functionality" you are referring to. I have not "implemented" anything. The window is just a standard JFrame with whatever that provides. There are various themes you can install and select, so maybe one of those is installed and changing it? Not using Windows I have no clue what you're on about.

majenkotech, 2018-10-06:
I think I have (now my Windows PC has finally been booted long enough for the mouse buttons to actually do anything when pressed ... jeez... how do you guys use this pile of crap?) found the menu to which you are referring - you mean a right click on the window icon in the task bar, yes?

UECIDE is a Java application. That one entry is all you get for a Java application. There's nothing you can do about it. It's Java.

You want it pinned to the taskbar then right-click the shortcut and pin it to the taskbar. There's absolutely nothing at all you can do from within a Java program to change a menu that only exists at the whim of the operating system it's running in.

Java is cross-platform. Platform- (and especially version-specific platform-) specific things like a context menu on a taskbar entry are nothing that Java ever even wants to consider touching.

closed 2018-10-06 by majenkotech

sensboston, 2018-10-06:
OK, thanks for explanation.

sensboston, 2018-10-06:
I found a simple workaround. If you not a principal Windows hater (BTW, for the real professional, OS shouldn't matter much), you may add it in FAQ section.

[image]

- create shortcut to javaw.exe on your desktop.
- add to the shortcut executable path ``-jar "C:\Program Files (x86)\Majenko Technologies\UECIDE\lib\uecide.jar``, so whole patch should look like ``-jar "C:\Program Files (x86)\Majenko Technologies\UECIDE\lib\uecide.jar"``
- change "Start in" to ``C:\Program Files (x86)\Majenko Technologies\UECIDE``
- change icon to ``C:\Program Files (x86)\Majenko Technologies\UECIDE\UECIDE.ico``
- pin shortcut to the taskbar
- delete shortcut

majenkotech, 2018-10-06:
Can't you just right-click on the icon in the start menu and pin to taskbar?

sensboston, 2018-10-06:
Yes I can, but in this case, if I run this shortcut, running java app will appear as an additional, separate icon on the taskbar, and Windows will not group these icons. Of course, it's not a big issue but if you're working on Windows, this is standard common programm behavior. 

P.S. @majenkotech, may I ask you to add ability to load last opened project to the IDE settings? It will be really good feature.

majenkotech, 2018-10-06:
It's already there. Check preferences (Editor -> Load/Save -> [X] Open last sketch automatically)

sensboston, 2018-10-06:
BTW, my solution doesn't work properly - UECIDE opened by this way, crashes immediately on chip programming step (don't know why, no error messages was received - just a simple crash).