UECIDE/UECIDE #111: ESP32 support

sensboston, 2018-10-07:
I'm looking for ESP32 support ('cause [Arduino Core for ESP32](https://github.com/espressif/arduino-esp32) is already exist) but don't know how to integrate this core to UECIDE. Any suggestions?

Also, for my needs will be nice to be able to save board/port to the project/sketch: I do have two projects for different boards, Arduino Nano and ESP32, but these projects should work together. Should I open a separate feature request for this?

majenkotech, 2018-10-07:
Porting a core is not a trivial operation. ESP32 support is something I have been considering, but I don't have any ESP32 boards to test with at the moment.  

As long as you're on the latest UECIDE you can add:

    #pragma board arduino-nano

to the top of your sketch.  You can do the same with `#pragma core ...` and `#pragma compiler ...` too if you need to select specific cores or compilers (usually just the board is enough since in most cases there is only one core and compiler associated with a board).

You can also use `#pragma option xxx = yyy` to set any of the options in the Hardware -> Options menu - there's even an option there to copy those settings to the clipboard in that format so you don't have to delve into the board / core configuration files to find the keys and values.

sensboston, 2018-10-07:
Where do you live? If in US, I can ship you ESP32 board, I do have a few extra (these boards are really cheap on AliExpress). Send me your address to [image]

Thanks for the tips. Hope, you'll add "port" setting to the #pragma option; if it has no sense for the current OS/board, UECIDE may simple ignore it ;)

majenkotech, 2018-10-07:
I want to overhaul the whole port thing. It's something I've tried before, but it hasn't been that great. 

I'm in the UK. I simply haven't got round to getting anything ESP32 yet (priorities...) but if you can recommend one I'll look at putting it on order...

majenkotech, 2018-10-07:
I also want to re-write the whole of UECIDE from scratch... there's still too many hangups from the old Arduino IDE getting in the way of things working smoothly.  Again... priorities...

sensboston, 2018-10-07:
I bought a few of [these boards](https://www.aliexpress.com/item/ESP-32-ESP-32S-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP32-Board/32826540261.html), they are working out of the box as a standard "ESP32 dev kit". If you don't wanna wait for delivery from China, you may buy [similar board on amazon.co.uk](https://www.amazon.co.uk/SeeKool-Development-Microcontroller-Integrated-Compatible/dp/B07DPP3BGZ/)

majenkotech, 2018-10-07:
I was wondering about maybe [this one](https://www.aliexpress.com/item/Lolin-ESP32-OLED-wemos-for-Arduino-ESP32-OLED-WiFi-Modules-Bluetooth-Dual-ESP-32-ESP-32S/32807531243.html) as well and use it for porting DisplayCore to the ESP32.

majenkotech, 2018-10-07:
Ok, both those are on order from China (I can get both from China for little more than one from Amazon, and I'm a cheapskate...)

sensboston, 2018-10-08:
OK, I'll follow UECIDE and hope you'll get luck porting ESP32 soon after order arrival ;)

P.S. Now I'm using Arduino IDE and UECIDE, but I like your IDE more (Arduino IDE is too primitive).

majenkotech, 2018-10-08:
That's exactly why I created UECIDE :)

majenkotech, 2018-10-15:
Well, my first ESP32 has arrived (the OLED one), and after a few hours of playing, I have at last managed to get the first compilation in UECIDE working on it.

The problem I have right now is that it uses a couple of python scripts.  One has an associated Windows binary, but the other one doesn't - so I need to find some way of running it in Windows - and that will mean having an installation of Python installed.

I have a "tools" system available in UECIDE for just this kind of thing, but I've never used it for something as complex as Python.  It's only been used for `ctags` so far...

sensboston, 2018-10-15:
Thanks for the good news! Hope to see a new release soon :) And I can be a beta-tester (and PR reviewer) for this feature.

P.S. I do have another feature requests for you, but will open a new issue.

majenkotech, 2018-10-16:
OK: I have released ESP32 support into the wild for testing. I have used it under Linux but not had a chance to test it under other operating systems yet.

There's only one board in there at the moment - just the "generic" one.  I'll add more boards later.

majenkotech, 2018-10-16:
Right - all the other boards are there now too. Give it a whirl and see what happens.

majenkotech, 2018-10-16:
Make sure that everything is updated - repos, families, etc.

sensboston, 2018-10-16:
OK, I updated repos and got ESP32 support working. It works well, I'm able to build my project without changes. Thanks a lot, good job!

P.S. I believe will be good to announce this feature (somehow) at the https://github.com/espressif/arduino-esp32

P.P.S. Now I'm waiting for the ``#pragma port `` feature ;)

majenkotech, 2018-10-16:
The port thing should be in the next release. It's done - I just need to test it thoroughly, and that means using it myself for a while.

closed 2018-11-04 by majenkotech