UECIDE/UECIDE #112: Visual Studio style shortcuts support and smart tabs

sensboston, 2018-10-15:
Will be really good to have some of VS keyboard shortcuts support in UECIDE. At least, Ctrl+K,C & Ctrl+K,U for comment/uncomment current line or selection. Full list of the keyboard shortcuts can be found [here](http://visualstudioshortcuts.com/2017/).

Another feature I'd like to see the UECIDE, it's a "smart tabs": when I press ENTER key, editor should position caret on the new line depending from the previous line (by inserting spaces and/or tabs), using C syntax. I.e. if I had bracket '{', new line for code input should be position of this bracket + tab.
The same behavior should be used if I press TAB key on the new line.

majenkotech, 2018-10-16:
I'm not sure it's possible to do those kind of two-character shortcuts. I used to have a menu entry for comment/uncomment - not sure where it has gone now.  I think I was planning on moving it into a plugin, but never got round to it.

I should put it back in again.

majenkotech, 2018-10-16:
As far as "smart tabs" goes - it already does that, doesn't it?

sensboston, 2018-10-16:
Nope, sir. At least not on Windows platform.

majenkotech, 2018-10-16:
The platform should make no difference - it's Java.  In the preferences (Editor -> Tabs) make sure you have enough "spaces for tabs" to make it worth while and noticable...?

sensboston, 2018-10-16:
"Spaces for tabs" isn't a "smart tabs" :) I already described this feature; it's will speedup code writing and it's an industry standard feature, most of IDEs has it.

majenkotech, 2018-10-16:
I'm not on about spaces for tabs. That converts tabs into spaces, which is completely different. The spinbox that sets the number of spaces not only sets the number of spaces to replace a tab with if that option is on, it also sets the width of a tab when it's a normal tab.

It defaults to 2, which is what the Arduino IDE uses (ugh!) but it's better to increase it to 4 or 5 so you can see your tabs better. Maybe also turn on the displaying of tabs as well, so you can see where they are.

sensboston, 2018-10-16:
My Windows 10 is booting about 10 seconds from the POST to desktop.
OK, I see what you mean, UECIDE has some auto-formatting feature; could you add a TAB key processing: if I pressed TAB before some text, it also should be moved according prev. line formatting (if any)?

majenkotech, 2018-10-16:
No, I don't think I can do that. I don't have that much control over the editor - it's a third-party plugin (RSyntaxTextArea).

sensboston, 2018-10-16:
It's OK, no worries.

majenkotech, 2018-10-27:
3a0c5be: Shift+Ctrl+C now toggles single-line comments for the current line or selection

closed 2018-10-27 by majenkotech