UECIDE/UECIDE #113: "Close Ctrl+W" (sketch) menu action causes app quit

sensboston, 2018-10-16:
"Close Ctrl+W" (sketch) menu clock causes app quit: probably, it's a bug.

majenkotech, 2018-10-16:
If you close the last window then yes it quits. That's how the Arduino IDE works too.

majenkotech, 2018-10-16:
On OS X it alerts you, since on that platform people expect the application to linger around with no window. On all other operating systems, when you close the last window of an application the application generally quits.

sensboston, 2018-10-16:
Arduino IDE shouldn't be a good example (to copy behavior). For me it was surprising and kinda illogical. I expected to get no project window, ready to open new sketch.

majenkotech, 2018-10-16:
But that's not how Windows applications work.

majenkotech, 2018-10-16:
But UECIDE isn't supposed to be like them. It's supposed to be like the Arduno IDE. It's a replacement for the Arduino IDE for people that need more from the IDE but don't want all the complexity of Atmel Studio, etc. And as such it has to work in the same way as the Arduino IDE so users of the Arduino IDE know what to expect.