UECIDE/UECIDE #114: Editor popup menu improvements

sensboston, 2018-10-17:
Please don't swear :) but could you add additional items to the editor popup menu? I would like to see two more items (context dependant):
- open file (at cursor position, for example at ``#include`` section) 
- go to declaration/definition (for variables, including external references)

I believe UECIDE become "IDE-ish" with these features, and it's not too hard to implement 'em ('cause you already have all references list, right?)

P.S. Will be good to add additional search results window (to the bottom pane) and "Find all references" to the popup menu, but this is a little bit harder.

P.P.S. If you found what I submitting too many feature requests, please let me know...

majenkotech, 2018-10-26:
How's this for you?

[image]

Can't really do the "open file" thing - it only makes sense for header files that are part of your sketch, and those you have in the tree on the left (and now in the Sketch menu as well).

majenkotech, 2018-10-26:
I've also created a "minimalist" mode:

[image]

sensboston, 2018-10-26:
Thanks! I'll try today or tomorrow and let you know.

majenkotech, 2018-10-26:
I haven't released it yet. It's in my *highly experimental* new version with many many changes to the way the UI works.

I guess I could build a pre-release version...

majenkotech, 2018-10-26:
Ok, there's a beta release now: https://github.com/UECIDE/UECIDE/releases/tag/0.10.0-beta1

sensboston, 2018-10-27:
New menu "go to" is working, I confirm. Commenting shortcut is not working (for me). Also, commenting one line is useless - it has a sense when you can quickly comment/uncomment whole selected block of code only; non-default (non Word Star compatible) shortcuts is also completely useless for me.

By the way, you should try (at least once) Visual Studio. It's an industry standard used for a decades around the world. Your own "bicycle invention" isn't too helpful, sorry.

P.S. I finally ended up with [Visual Micro](https://www.visualmicro.com/): it has (mostly) all features I'm using and used for a years (because it's a plugin to VS). It also has a nice ability to keep and properly handle projects for different platforms in one solution. And it's also free (but not an open source) however I paid 'em (to remove nag screen :) ).

Sorry for disturbing you! I really appreciate your work but unfortunately UECIDE can't compete with Visual Studio, it's a different kind of software, practically.

majenkotech, 2018-10-27:
I have been working on the other things today.  In the next release, comments will be toggled with ctrl-shift-c and work either on the current line or the selected text. If the first line is commented then comments are removed for any selected lines. If the first is not commented then comments are added to all selected lines that don't have comments.

I can't (and won't) use VS since I don't use Windows. I do have a copy somewhere around, but never use it unless I have to write a piece of Windows GUI software (and then under duress) - so that means never.  And UECIDE is not supposed to compete with VS. As you say - it's a different kind of software. 

* Arduino IDE: Robin reliant
* UECIDE: Volvo S90
* VS: MiG-29

I have found a way of doing the "open include file" thing - I have been playing with the token parser of RSyntaxTextArea. I've added all sorts of things now: open a header file if it's part of the sketch, open the library folder if it's not part of the sketch, go to the manual page for a standard Arduino function, go to the definition of globals and defines, etc. All sorts of funky things.

* c35acd6: Advanced token parser allows more context-aware options in popup menu

closed 2018-10-27 by majenkotech