UECIDE/UECIDE #117: Add MQTT control panel

majenkotech, 2018-11-15:
It could be quite fun and useful to have an MQTT control panel built into a project. Maybe also provide a standalone version to run the same configuration files outside UECIDE - so make it a separate project that is embedded (an editor?) in UECIDE like the graphics editor is.

majenkotech, 2019-01-18:
Thus is just a record of an idea is had. Whether it gets implemented is another matter. If it did it would be coupled with a standalone program and an Android app, so would be a lot of work. It probably will never be done, but the idea is there if it ever feel the need.

SantoRemedio, 2019-01-22:
A MQTT "message capture" (a la wireshark) will be nice as a debugging aid. While I can log messages via command line app, that doesn't provide time stamp or message selection based on conditions. I haven't seen anything like that in market. It guess it's easier to do than a full fledged MQTT client and provide a functionality not available otherwise.

SantoRemedio, 2019-01-22:
Yes, that is. With options to save/reload the capture data. Maybe a plugin ...

majenkotech, 2019-01-22:
Probably won't be able to do it 100% as a plugin. The client itself would probably need to be written in Java.