UECIDE/UECIDE #121: Error compiling valid program

SantoRemedio, 2019-01-17:
**Describe the bug**
Compilation fails with error:

> • Error at line 6 in file testUE_proto.h:
> ‣ expected constructor, destructor, or type conversion before ';' token

*testUE_proto.h*
```
#ifndef _UECIDE_FUNCTION_PROTOTYPES
#define _UECIDE_FUNCTION_PROTOTYPES

// This should be inserted at line 3

displayTemp(float t1, int t2, float d);
void loop();
void setup();

#endif
```
**To Reproduce**
Compile this code
```
void displayTemp(float t1, float t2,  float d);
void setup() {}
void loop() {}
void displayTemp(float t1, int t2,  float d) {}

```

**Desktop (please complete the following information):**
 - OS: Linux Mint 19
 - UECIDE v. 0.10.2, Arduino Nano w/ATmega328

**Additional context**
Compile OK with Arduino IDE 1.8.7

majenkotech, 2019-01-17:
It's obviously losing the void when making the prototype. It could, i guess, be the additional manual prototype that is throwing it out...

majenkotech, 2019-01-17:
No, believe it or not it's two spaces before the second `float` in `displayTemp` that seems to be causing it.  That's very odd.

majenkotech, 2019-01-17:
Ok, this is a strange thing with ctags. Multiple spaces in the prototype are compressed into single spaces in the signature. 

    displayTemp	/tmp/untitled43/build/untitled43.ino	/^void displayTemp(float t1, int t2,^^float d) {}/;"	f	line:4	signature:(float t1, int t2,^float d)

And because of that it can't locate the signature in the prototype.  That's going to have to be manually worked around...

SantoRemedio, 2019-01-17:
> No, believe it or not it's two spaces before the second `float` in `displayTemp` that seems to be causing it. That's very odd.

Right. Tested it with example & real pgma.

majenkotech, 2019-01-17:
Ok, a simple regex to compress \s+ into " " seems to fix it.

Fixed in 7331199

majenkotech, 2019-01-17:
I am releasing 0.10.4 with this fix in. I deemed it worthy of a release all of its own ;)

closed 2019-01-17 by majenkotech