UECIDE/UECIDE #123: Search text across all project files

SantoRemedio, 2019-01-19:
**The problem**
Due to the limitations of Arduino platform, sketches tend to have lot of global variables used across multiple source files.

When writing/debugging sketches, you frequently need to search for all uses of a given symbol/text. As today, that must be done by invoking the same **Find** command in every possible source file. Too much work.

You can also open a new window and run *grep text* in the command line, but it's not very convenient.

**The Solution**
At the **Find** menu, add the option to search a text across all project files, like Arduino IDE.

majenkotech, 2019-01-19:
The option exists in the tree panel, but it am not sure it actually works at the moment. It's something I need to rewrite from scratch.

majenkotech, 2019-01-19:
I was right - it no longer works.  So I have written a Javascript plugin that does the job.

It requires the latest version of UECIDE (or a very recent one anyway) for the plugin to operate correctly, but it adds an extra "search" tab to the left-hand panel.  It's pretty simple at the moment, but being a Javascript plugin extra functionality can be added really easilly.

majenkotech, 2019-01-20:
Ok, I've added some more enhancements to search now.  You can:

* Move the tab to different locations in Preferences if you don't like it in the left panel.
* Do case sensitive / insensitive searches using a checkbox in the tab panel.

If you want to install the plugin and have a play and let me know if there's anything else it can do that'd be great. Otherwise I'll put this one to bed.

SantoRemedio, 2019-01-20:
Works like a charm.

Wish list:
- Whole word.
- Regular expressions.
- Sometimes I like the "list" style of result (with links to the source text); others I like just jumping from one hit to another (like now). Would be very nice having both options available.

majenkotech, 2019-01-20:
I have made it now so that when you do a search it automatically goes to the first result. If you repeat the same search (either by pressing the search button or by hitting enter) it goes to the next result for you. And loops when it gets to the end.

closed 2019-01-31 by majenkotech