UECIDE/UECIDE #124: Modal dialogs turns UECIDE unresponsive to keyboard/mouse events

SantoRemedio, 2019-01-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

majenkotech, 2019-01-20:
> Click on any part of the main UECIDE window underneath (to cover the dialog with the main window).

I'm not sure I follow. That is not possible. The dialogs are application modal and block everything. You can't interact with the main app window while an application modal dialog is open. If the main window can become placed in front of the dialog then that is down to a fault with your window manager (or its interaction with Java).

majenkotech, 2019-01-20:
Which WM are you using with Tara? Cinnamon, MATE or Xfce?

majenkotech, 2019-01-20:
That link is broken.

Opening a modal dialog *will* block all input to the main app. That is what a modal dialog *does*.  It should be impossible to get the main app window to end up in front of the modal dialog window. If it can happen then it's not UECIDE that is letting it happen. 

I am downloading Tara MATE now to try it out.

SantoRemedio, 2019-01-20:
Duh! Privacy settings defaults to private.

Try it now: https://youtu.be/_I_rGpMMXqI

[Window Manager Mate 1.20.1 (stock installation).]

majenkotech, 2019-01-20:
That certainly should not be happening. And it's not something I have seen happen, and can't replicate on any other system.  Mint has nearly downloaded and I will set up a VM to test it out.  

There should be no way at all that the window manager should allow the dialog to get behind its parent window.

majenkotech, 2019-01-20:
Ok, I replicate it on Mint.  It looks like something that it doesn't like with the custom window rendering of JTattoo.  Nothing I can do about that, it's not my code (and I have no clue how it works).  If you change the window style to "Gnome" in the theme preferences the problem rectifies itself.

To do it outside of UECIDE you can edit ~/.uecide/preferences.txt and look for `theme.laf=...` and set it to `gnome`.

majenkotech, 2019-01-20:
Right, yes it's the custom decorator.  I did have code in there to allow it to be disabled, but for some reason it wasn't working. (Actually that "some" reason was that it was being run too late to take effect).  I have fixed that code now and it honours the setting.  It should default to disabled which will fix the problem, but the theme won't look *quite* as good as it would since it will now have the system default window decorations. 

But at least the choice will be yours: custom window decorations and possible dodgy modality management, or system window decorations and fixed modality management.

majenkotech, 2019-01-20:
"Fixed" in e85abab

closed 2019-01-20 by majenkotech