UECIDE/UECIDE #126: Pause button for serial terminal

SantoRemedio, 2019-01-31:
**Is your feature request related to a problem? Please describe.**
**`Serial Terminal`** is useful for debugging purposes, but the inability to pause the output to examine data make it less useful.

**Describe the solution you'd like**
Add a *pause* button to the serial terminal to pause the output from program.

**Describe other alternatives**
... Unplugging the USB cable ... not good.

majenkotech, 2019-01-31:
I'm not sure about a "pause" button, but a button to toggle the auto-scroll-on-reception is doable.

majenkotech, 2019-01-31:
Actually, the way that the serial terminal widget works, disabling the auto scroll doesn't work. It relies on the scrolling to create room to place new text. It's a bit crap like that.  I can pause all reception (discarding data), or I can have it pause the display and all incoming data overwrites the last line. I can't have it pause just the scrolling and have the text append off the bottom of the window.

SantoRemedio, 2019-01-31:
Use case: debugging a program with lots of Serial.print everywhere. Me, waiting for some specific text to appear (the bug). I want to stop the scrolling to be able to inspect the history leading to the bug. Whatever comes after the bug, is irrelevant.

SantoRemedio, 2019-01-31:
I like the "discard lines while paused": you can suppress tracing irrelevant code while reaching the section you are debugging.

majenkotech, 2019-01-31:
Great. That's the easier option to implement.

majenkotech, 2019-01-31:
Added in 9359928

closed 2019-01-31 by majenkotech