UECIDE/UECIDE #127: Clear button for Serial Terminal

SantoRemedio, 2019-01-31:
**Is your feature request related to a problem? Please describe.**
I want to clear the `Serial Terminal `window before starting a new run or after some event in the program, to avoid confusion between old and new data displayed.

**Describe the solution you'd like**
A `Clear` button in the `Serial Terminal` window that erase all content.

**Describe alternatives you've considered**
Closing and reopening the `Serial Terminal`, but that means a program reset. Not good when in the middle of something.

majenkotech, 2019-01-31:
Also an option in Preferences to auto-clear when you open the window, or when the port is opened could be good.

majenkotech, 2019-01-31:
Added in 9359928

closed 2019-01-31 by majenkotech