UECIDE/UECIDE #130: Editor shift cursor by +1 after left/right, up/down keyboard sequence

SantoRemedio, 2019-02-02:
**Describe the bug**
While editing, if you move the cursor left or right and then up or down, the cursor ends one column off to the right.

**To Reproduce**
1. On a new empty sketch, write this two lines:
    FIrst line: "ABCDE". 
   Second line: "12345"
2. Click on the first line, first column, left to the "A". Cursor is now between "A" and "B".
3. Press _right_ cursor, then _down_ cursor. 
4. Cursor is now between "2" and "3" in the second line.

**Expected behavior**
Cursor between "1" and "2".

**Screenshots**
This video shows the problem: [Cursor movements](https://youtu.be/B5PvAk3s2_w)
All cursor moves by keyboard.

**Desktop (please complete the following information):**
 - OS: Linux Mint 19 (updated).
 - Uecide 0.10.5

majenkotech, 2019-02-02:
Actually, that might *not* be the problem. It's something that needed doing anyway though. I am still seeing the issue. It loves to be on the second character of a line, doesn't it?

majenkotech, 2019-02-02:
Ok, I can see the cause - though not the solution yet. It's the margin setting of the editor. If you set a left margin it screws up the cursor positioning. I suspect that's a bug in RSyntaxTextArea, which is not something I have much control over. I must see about porting the latest version of that widget over to UECIDE and see if that improves things.

SantoRemedio, 2019-02-02:
Yes. I'm using Bitstream Vera Sans Mono,bold,16, with 14 for left margin. I checked, and any value for left margin over 4 triggers the problem.

My setup is optimized for legibility and visual confort, which explains the use of non-default options.

As for the margin, I find visually uncomfortable the code right next to a vertical black line:
[image]

vs using 14 for the left margin:
[image]

easier on my eyes.

majenkotech, 2019-02-02:
Absolutely. The left margin is a needed feature that I really don't want to lose.  I just did a test upgrade to RSyntaxTextArea without my custom modifications (which alas I can't seem to find in a source code form :( ) and it didn't fix the problem. So maybe it's not the editor itself at fault but how I'm using it - or maybe it's just plain broken and there's not a lot I can do about it...

majenkotech, 2019-02-02:
Ok, I think I have it.  I was using `setBorder()` to set an empty border on the text area. However there is a `setMargin()` that creates an empty margin. Using that seems to fix the problem.

majenkotech, 2019-02-02:
Fixed in 7717841

closed 2019-02-02 by majenkotech