UECIDE/UECIDE #132: Artistic Style Plugin fails

SantoRemedio, 2019-02-08:
**Describe the bug**
The Artistic Style (astyle) plugin fails to execute.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Tools'
2. Click on 'Auto Format with Artistic Style'

The error message is:

> Unable to start process
> Cannot run program "/tmp/astyle" (in directory "/tmp/build-2cbcb34e-2679-41b3-9134-2657b3f424f6"): error=2, No such file or directory
> 

The /tmp directory has this file:

> rand@Gear:/tmp$ ls -lh astyle
> -rwxr-xr-- 1 rand rand 366K Feb  8 08:22 astyle 

> rand@Gear:/tmp$ file astyle
> astyle: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.24, BuildID[sha1]=d716cf952098469a17f366969e4445e006253125, stripped
> 

This file is being copied and deleted by UECIDE.

and this directory:

> rand@Gear:/tmp$ ls -lhd /tmp/build-2cbcb34e-2679-41b3-9134-2657b3f424f6
> drwxr-xr-x 2 rand rand 4,0K Feb  8 08:32 /tmp/build-2cbcb34e-2679-41b3-9134-2657b3f424f6
> 

If you attempt to execute directly:

> rand@Gear:/tmp$ ./astyle
> bash: ./astyle: No such file or directory
 
Weird. Why _No such file_ ?

However, the same scenario works if I use _astyle_ from Linux repositories.
[These tests where done after uninstall _astyle_ as plugin and as Linux package, then reinstall _astyle_ plugin only.

**Expected behavior**
The code reformated.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Linux Mint 19 Tara 64 bits (updated).
- MATE 1.20.1
- Uecide 0.10.5.
- Astyle plugin 1.0.4

majenkotech, 2019-02-08:
The Artistic Style "plugin" you are using is deprecated. There is a Javascript one in the plugin manager that replaces it. It is a two-part one - a JS plugin that provides the interface, and a "tool" package that provides the executable.

It sounds to me like you may have /tmp mounted as a separate filesystem (tmpfs?) with execute disabled on the mount. That becomes "not a problem" with the newer JS plugin.

SantoRemedio, 2019-02-08:
I don't see that option in my updated plugin list:
[image]

[Yes, my home directory volume is encrypted and mounted by hand during boot time.]