UECIDE/UECIDE #137: Multiple tabs alternative navigation

SantoRemedio, 2019-03-11:
**Is your feature request related to a problem? Please describe.**
When dealing with multiple files opened (in multiple tabs), selecting a particular open one is difficult because you need to navigate across tabs using only "prev" and "next" buttons, multiple times, without a hint of where is that particular tab located (to the right or to the left), because tabs doesn't fit in the panel space. This is cumbersome and slow.

[image]

**Describe the solution you'd like**
Replace the "prev" and "next" buttons with a drop box scrollable list showing all open tabs.

[image]

**Describe alternatives you've considered**
Using the Project pane for selecting file/tab. This is not optimal, because you need to double click on the file name, which expand/collapse the tree at that point, which is an unwanted side effect.

majenkotech, 2019-03-11:
Unfortunately I don't get that sort of control over a jtabbedpane. It's not possible to add a menu like that as far as I know. However I could add a mouse listener and create a right click popup menu on the tabs themselves which lists all the tabs. Would that be a suitable work around?

SantoRemedio, 2019-03-11:
That's a nice and convenient solution.

majenkotech, 2019-03-12:
Some themes already provide that extra menu button. Not all, but some. It's down to the theme to provide it, not the software. 

However I have added a right-click menu anyway as an alternative way of doing the same thing - so whatever theme you have selected it should make navigating tabs easier.

closed 2019-03-12 by majenkotech