UECIDE/UECIDE #138: Upload precompiled binary

SantoRemedio, 2019-03-12:
**Is your feature request related to a problem? Please describe.**
I want to save the current working executable version while developing a new one, to quickly reinstall the working one without needing the original source code.

**Describe the solution you'd like**
An new option: _Sketch->Upload precompiled_. This allow to select a file and sending it to Arduino. It will work like _Re-upload last compiled_, but asking for the file first.

**Describe alternatives you've considered**
Using _avrdude_ directly or _Xloader_, but they are not user friendly and require switching to CLI.

**Additional context**
I think this is a common problem when you have only one hardware instance of your project and you need to make a major upgrade. You use the same machine for testing and production, and you want to make it quick and safe.
I tried to solve it using Git branching, but after adding a few new files, branching back and forth to _master_, proves to be not a quick task.

majenkotech, 2019-03-12:
It's a plugin I have been needing to write for a while, yes. It is on the cards.

majenkotech, 2019-03-12:
We'll keep it open as a reminder to me.