UECIDE/UECIDE #139: File changed indicator/action

SantoRemedio, 2019-03-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

majenkotech, 2019-03-16:
I used to have indication of changes since last save. Not sure how possible an indication like that per file for changes since last compile or upload.

I could always have the compilation routine modify your source code for you.  Have it look for a comment starting

    // $Compiled: 

and add the compilation time and date to the end of it:

    // $Compiled: Sat, 16 Mar 2019 20:08:32 +0000

SantoRemedio, 2019-03-16:
"Changed since last save" is OK. My setup saves the source files before compiling, anyway.

I don't think "$Compiled" is a practical idea. It require for me to visually inspect each open tab to check if the file has changed.  And it's not clear the semantic when you mix .ino, .cpp and .h. What happens when a globally used .h changes?

majenkotech, 2019-03-16:
Changed since last save is on my list of things to reinstate. 

At the moment the system's not clever enough to know to recompile source files if an included header has changed. It would be nice if it was, but that would be a mammoth task to implement and mess up the whole of the existing compilation structure.

majenkotech, 2019-03-29:
As part of my revamp of the UI, and the adding of a new LAF (Material look) I have some interesting chages.  For instance, the tab now has a modified warning icon. Things have been tidied up a little, and I've just tried to make it more "refined"...

[image]

Thoughts?

SantoRemedio, 2019-03-30:
Very nice look!

Personal preferences:

1) "Cancel" instead of "Don't close". One word vs. two (or three).
2) "Save" and "Discard". One word, both meaning positive actions (save, discard) rather than absence of action (negative action).
3) Button order: "Discard", "Cancel", "Save". Because "Cancel, Save" at the right is the most generic and consistent way to arrange any kind of dialog, with extra options to the left. 

There is no standard in Linux about the button order or action semantic. Standards are different between Windows and Mac. That's a problem for a multi platform product like UECIDE. Must be dialogs adaptative to platform guidelines?

I like the MacOS user interface guidelines: 

> A button that initiates an action is furthest to the right. The Cancel button is to the left of this button.

[image]

Which is consistent with Android guideline:

> The dismissive action of a dialog is always on the left. Dismissive actions return to the user to the previous state.
> 
> The affirmative actions are on the right. Affirmative actions continue progress toward the user goal that triggered the dialog.
> 

For more details, look at this: [OK/Cancel on left/right?](https://ux.stackexchange.com/questions/1072/ok-cancel-on-left-right)

In Linux, no standard or convention whatsoever, but pure and simple chaos. Here is a sample of some apps I use:

### Arduino IDE
[image]
Button "Yes" means "save".

### PulseView
[image]
Button "Yes" means "discard".

### Netbeans
[image]

### Libre Office
[image]

### Shotcut
[image]

majenkotech, 2019-03-31:
Here it is reworded and reorderd - also with the new Arduino theme I am working on (ghastly, but people seem to like the way the Arduino IDE looks... weirdos...)

[image]

SantoRemedio, 2019-03-31:
For me is good enough, but one more suggestion:

You can add file/sketch to the title bar ("unsaved file - . So, after a time, a UECIDE daily user just look at the title bar, which contains all the needed information. 

(I just photoshopped your dialog. It may not be evident at first sight)
[image]

If file/sketch name is too long, it can be truncated, because it's in full in the message body.

majenkotech, 2019-03-31:
Sure, that's simple enough.
[image]

closed 2019-04-05 by majenkotech