UECIDE/UECIDE #140: Add file name/project name to Unsaved File dialog

SantoRemedio, 2019-03-17:
**Is your feature request related to a problem? Please describe.**
When quitting UECIDE, a dialog is shown if there are unsaved files.
[image]
The dialog doesn't name the file, nor its project. That information is needed to decide what to do next.

**Describe the solution you'd like**
1. Modify the dialog to show the file and project name.
2. Bring the project window and file tab to the the front, underneath the dialog.
[image]

**Describe alternatives you've considered**
None.

majenkotech, 2019-03-17:
I can't get the tab or window to focus, because it all happens in the event handling thread and that thread needs to finish the current iteration before the swing thread can process any requests to modify the window. It's a problem of blocking dialog boxes. Not a lot I can do about that. However I have got it to show the program and file name in the dialog.

SantoRemedio, 2019-03-17:
I can suggest two variants:
- a single dialog listing all unsaved files, with buttons "Cancel" and "Save All".
- a dialog for each file, with buttons "Cancel", "Save" and "Save All". It this case, the IDE stop asking any more question.

majenkotech, 2019-03-25:
I already changed the wording of the buttons. This is how the close dialog looks at the moment:

[image]

majenkotech, 2019-03-25:
I'm also considering a dialog box a bit like the Gimp one.

SantoRemedio, 2019-03-25:
I like the Gimp dialog better. It works with multiple unsaved files, because it's only one dialog and it shows all unsaved files at once.
[image]
 
For the one-file-save dialog, I will propose a change in name font/style/size/color, to make it stand out.

majenkotech, 2019-03-25:
The problem I have with the Gimp-style box is I'd like one box that covers all sketches if you do a quit. However that's not possible with UECIDE - it just sends a "quit" signal to each window in turn, and it's up to that window to ask if it should close or not.  So you could still potentially get multiple dialogs. However it would only be one per UECIDE window, so not *too* bad...

majenkotech, 2019-04-05:
I'm going to keep it as it is for now. It's improved, with the new wording, but

closed 2019-04-05 by majenkotech