UECIDE/UECIDE #141: Sketch name in Serial Terminal window bar

SantoRemedio, 2019-03-20:
**Is your feature request related to a problem? Please describe.**
Currently the Serial Terminal put the device id (like /dev/ttyUSB1) in the title bar. When working with two or more boards, it's not easy to determinate to which board it's attached to what Serial Terminal window.

[image]

**Describe the solution you'd like**
Put the name of the sketch in the window title, like
[image]

**Describe alternatives you've considered**
Using an external terminal emulator to avoid confusion.

majenkotech, 2019-03-24:
Not so sure about that one.  The terminal isn't really associated with the sketch. It's associated with the device that is selected in the current window. And if you open multiple windows (as is the future plan) on different ports from within the same sketch window the sketch name would be irrelevant, since you're not just looking at the one device.

Better may be to display the USB or associated board information in the titlebar so you can better see which device you are talking to.

majenkotech, 2019-03-25:
Added in b19c11c

closed 2019-03-25 by majenkotech