UECIDE/UECIDE #142: Make Serial Terminal independent

SantoRemedio, 2019-03-22:
**Is your feature request related to a problem? Please describe.**
Currently Serial Terminal launches only one time per UECIDE instance, even if there are several serial devices connected. 
If you want to connect to two devices, you have to launch another instance of UECIDE or call a terminal emulator of your choice.

**Describe the solution you'd like**
When you click on Serial Terminal and there is two o more serial devices connected, a dialog must be presented to let the user choose which port connect to.

**Describe alternatives you've considered**
Launching another instance of UECIDE, which clutters the desktop.
Using another terminal emulator. I tried several, but none is really user friendly ... they want _parameters_ ... 

**Future development**
This open the possibility of using only one Serial Terminal Windows for both devices. The output from each device can be in a different color, intermixing complete lines from both devices according to arrival time.
That way you can easily debug any inter-device communication, because messages will be in the correct time sequence.

majenkotech, 2019-03-24:
At times like this I usually launch Minicom. I have been considering ways of wrenching the serial terminal away from the sketch window.  Traditionally the serial terminal is associated with the device that you have programmed - however this is less and less the case, especially with devices that aren't programmed through a serial port. The serial port is the same one you program the device through, and that is increasingly less correct nowadays.

I am considering having it so if you left click you get the current system - a terminal that opens the currently selected device's serial port, but if you right click (or CTRL-click for those mac users that can't handle two mouse buttons...) it gives you a popup menu with all the serial ports listed for you to select one to open a new terminal on.

SantoRemedio, 2019-03-24:
You only can launch Serial Terminal from a sketch. Sketch has a name. The sketch name is more relevant that device name.

majenkotech, 2019-03-24:
It seems to me the two concepts are mutually exclusive. Either you have one serial terminal per sketch window, which is associated with the sketch, or you have multiple serial terminals launchable from each sketch window, which are associated with whatever device it is you are communicating with. Which is it you want?

SantoRemedio, 2019-03-24:
I want the sketch name on the Serial Terminal window title, instead of/in addition to device name (to keep everyone happy).
About the other things, they are really out of scope here, now I know. What I want is a centralized logging service for multiples USB-connected devices (Arduinos and others). The idea is to produce an output log with messages in arrival time sequence (timestamped & color coded). Hacking that inside Serial Terminal is non sense, it really has to be something different.

majenkotech, 2019-03-24:
Since your other comments I have been pondering the creation of a separate serial debugging system. One where you can attach / detach multiple devices, have them identified, have timestamps, etc. Even maybe some decoding of common protocols (MODBUS for example).

majenkotech, 2019-03-24:
Added in b19c11c

closed 2019-03-24 by majenkotech