UECIDE/UECIDE #143: Serial Terminal Pause Status Indicator

SantoRemedio, 2019-03-24:
**Is your feature request related to a problem? Please describe.**
The button Pause for the Serial Terminal shows minimal differences between paused/normal state, and no hint about what status it is.

Normal status:
[image]

Paused status:
[image]

**Describe the solution you'd like**
Change the _Pause_ button to a _Play_ button when paused.

majenkotech, 2019-03-24:
Yes, that thought did cross my mind while creating that functionality. It is on the cards, but at a lower priority than just creating the functionality.

majenkotech, 2019-03-25:
Added in ae7242d

closed 2019-03-25 by majenkotech