UECIDE/UECIDE #146: Declaration mysteriously changed

SantoRemedio, 2019-04-21:
**Describe the bug**
A function declaration changed during the compilation process, rendering the sketch invalid. Same sketch compiles without error using Arduino IDE.

The declaration is:
`char *uptime2Str();`
and was changed to
`char uptime2Str();`

Grep over the sketch directory:
```
candid@Gear:~/Arduino/PowerSwitch$ grep uptime2Str *
grep: data: Is a directory
Display.ino:    u8g2.drawStr(70, 48, uptime2Str());
PowerSwitch.ino:char *uptime2Str();
PowerSwitch.ino:  client.publish(MQTT_UPTIME, uptime2Str(), true);
PowerSwitch.ino:char *uptime2Str()
```
The uecide reported error:
```
• Error at line 40 in file PowerSwitch_proto.h:
‣ new declaration 'char uptime2Str()'

• Error at line 92 in file PowerSwitch.ino:
‣ ambiguates old declaration 'char* uptime2Str()'
```
[image]

Content of PowerSwitch_proto.h
```
#ifndef _UECIDE_FUNCTION_PROTOTYPES
#define _UECIDE_FUNCTION_PROTOTYPES

// This should be inserted at line 163

void debug(const char *msg);
void debug(unsigned int msg);
void debug(unsigned int msg, unsigned int mode);
void debugln(const char *msg);
void debugln(int msg);
void debugln(unsigned int msg);
void debugln(unsigned long msg);
void digitalClockDisplay();
void setDebug(boolean b);
void setupSyslog();
void callback(char *topic, byte *payload, unsigned int length);
void fcnBadValue();
void fcnDebug(char *payload);
void fcnProgram(char *payload);
void fcnPumpOn(char *payload);
void fcnRelay(char *payload);
void fcnReset(char *payload);
int parseIntPayload(char *payload);
void setupMQTT();
unsigned long ntpUnixTime();
void checkNetworkServices();
unsigned long getMillis();
void loop();
void pumpOn();
void readSensor();
void reset();
void setup();
void setupCron();
void setupPump();
void setupRTC();
void setupSensor();
void setupTimer();
void setupWiFi();
void syncTime();
char uptime2Str();
```
#endif

**To Reproduce**
I'm afraid I need the whole source code to reproduce this.

**Desktop (please complete the following information):**
 - OS: Linux Mint Tara 19
 - UECIDE 0.10.6
 - Arduino IDE 1.8.9

**Additional context**
All software updated.
The error persisted after purging old output file and after restarting UECIDE.

majenkotech, 2019-04-21:
It's losing the * when creating the prototypes. Not sure why...

majenkotech, 2019-04-21:
It's associating the `*` with the function name, not the return type. A problem with my parser function. A simple stop-gap fix:  add a space to the actual function implementation:

    char * uptime2Str() {
        // ...
    }

SantoRemedio, 2019-04-21:
That compiled OK!

majenkotech, 2019-04-21:
I've added a special catch to the prototype generator to check for a * at the start of the function name.

Fixed in b72a51e