UECIDE/UECIDE #147: Unable to upload to ESP8266 via CP2104 due to COM port detection error

Kashouryo, 2019-06-01:
**Describe the bug**
UECIDE cannot find the correct COM port that CP2104 use. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Plug an ESP8266 with CP2104(2102 might also work) as a programmer
2. Select Generic ESP8266 as the board
3. However, under the port section, UECIDE cannot find the correct COM port(my Arduino IDE displayed COM19, but UECIDE only have COM32 available)
4. If the user keeps going and click upload, the upload will fail due to UECIDE cannot find the correct COM port

**Expected result**
The sketch can upload normally

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10 v1803
 - UECIDE: 0.11.0-beta1
 - Version [e.g. 22]

majenkotech, 2019-06-01:
UECIDE can only report the COM ports that are handed to it by JSSC (Java Simple Serial Connector). If it's getting the wrong list, that's not UECIDE's fault, but JSSC's fault.  JSSC is already at the latest version (2.8.0) and that hasn't been touched for a number of years (released 2014).

JSSC neither knows nor cares what the COM port is - it just reports what it thinks is there. I have never seen, nor heard, of a problem where JSSC reports the wrong COM numbers. It has always reported just what is on the system. There's no way AFAICT that it can possibly just make stuff up.

Maybe your registry is corrupt or has some other invalid information in it pertaining to COM ports.

Kashouryo, 2019-06-02:
It looks like JSSC does not like CP2104... I tried my Arduino UNO with ATMEGA16 as the USB controller, and it works fine. Thanks for the information!

closed 2019-06-02 by Kashouryo

Kashouryo, 2019-06-15:
I recently found an alternative library that still under maintenance called jSerialComm. It looks like a nice replacement for the JSSC library.
here is the link if you want to check it out https://fazecast.github.io/jSerialComm/

majenkotech, 2019-07-09:
I have replaced jSSC with JSerialComm now. Version 0.11.0-beta-3 is the test version. It also comes (in the "full" windows versions) bundled with JDK 11. It was this need for display scaling in Java 9+ that prompted the change.  It'd be great if you could give it a test for me to see if it now fixes your problem.

Kashouryo, 2019-07-28:
Sorry for the late response. JSerialComm worked like a charm! The opening speed of the serial monitor is a bit slow(known issue). Besides that everything seems working fine. Thank you!

majenkotech, 2019-08-31:
I have finally got round to curing the slowness of jSerialComm. It seems the code is liberally littered with manual delays (`Thread.sleep(...)`) for no real reason that I can discern. I have removed them all, and I'll see if it has any negative effects.