UECIDE/UECIDE #148: How do you add a core that's not in the core repository?

Kashouryo, 2019-08-24:
There is an extremely cheap AVR compatible called LGT8F328P. And I'd like to add this core to my UECIDE. However, I cannot find any documentation on how to add the user's own core. Here is the GitHub repo of the core: https://github.com/LGTMCU/Larduino_HSP
This core might be a little different since it's also using the Arduino's built-in AVR-GCC. The core is licensed under the MIT license.

majenkotech, 2019-08-24:
Adding a core is a long involved problem. The issue is that UECIDE works in a fundamentally different way to the Arduino IDE - it splits out the compiler from the core, so the core has to conform to not only what UECIDE internally expects, but also what the compiler's configuration expects.

When you have a core that is based upon another core it's probably easiest to take the existing core that it's based on (the Arduino core in this case) and replace the folder with all the API code in it (the `arduino` folder in this case) and make any required changes to the `core.txt` file that are needed to support the new core.  This may be as little as changing the `name=` (which must be the same as the folder the core is in) and `description=` entries - or it may be more involved needing to change compilation flags (some of which are currently provided by the compiler, not the core) and maybe even adding extra code to be compiled separately.

Kashouryo, 2019-08-24:
Wow, that sounds complicated. Is it possible to change the subject of this issue to "Add LGT8F AVR support"? Because I am having no clue about what should I do to add this core.

majenkotech, 2019-08-24:
Alas, yes it is complex. I would love to be able to support Arduino JSON packages, but the whole way that UECIDE works prevents that really happening. You may ask "why did you do it differently to Arduino?" and the answer to that is: I didn't. Arduino did it differently to me. UECIDE's multiple cores / boards / compiles and the whole Plugin Manager pre-dates Arduino's Boards Manager by a number of years. And although it's far more complex I wouldn't change UECIDE's method now for all the tea in China. Without it I wouldn't be able to offer such things as avr-gcc 9.1.0 as an option for compiling with the standard Arduino core - something Arduino will never be able to do ;)

majenkotech, 2019-08-24:
Ok, I have attempted to create packages for it. It *seems* to be straight forward enough, just using a direct copy of the Arduino configuration with a few tweaks.  I can't guarantee that it'll work though...

Kashouryo, 2019-08-25:
Thanks for adding it in the board manager! At first, I can not successfully compile at first try. However, I did manage to compile with UECIDE with a minor action successfully.
[image]
In the pins_arduino.h, there is an include that uses an extra folder called standard. After I moved this folder to the LGT's boards folder, it compiles successfully(and fast!).
[image]

majenkotech, 2019-08-25:
That folder should be included in the core. Maybe the packaging script missed it...

Kashouryo, 2019-08-25:
There's one more thing that worth mention. 
In core.txt if I am using Arduino's core.cppflags and core.cflags, the LGT core lib will build successfully. The configuration that comes with the LGT package doesn't work.
[image]

majenkotech, 2019-08-25:
I thought I'd taken the LTO stuff out of there.  It's moved into the compiler package instead of the core, since not all versions of the compiler like it.

majenkotech, 2019-08-25:
I've uploaded 0.0.6 to the repo with the LTO stuff removed.  If you want LTO you'll need compiler 5.4.0 from the testing repository.  Soon that will enter the main repository anyway.

Kashouryo, 2019-08-25:
I just tested the 0.0.6 version on a real Arduino Mini(since currently I only have unsoldered LGT8F boards and my sodering iron was broken). And it works!(besides some intended minor issues since ATMEGA328P is not LGT8F328P)

majenkotech, 2019-08-25:
Fantastic!

Kashouryo, 2019-09-01:
Hello! I just tested with a real Larduino. It works fine with LGT8F328P(which is the module I have)! However, when I selected LGT8F328D-SSOP20, UECIDE throws some error and failed to compile.
But LGT8F328D is some what rare so I think this is good enough?

Here's the compiler error:

Compiling...
• Preprocessing...
• Converting binary files
• Compiling sketch...
◦ Fade_combined.cpp
• Compiling core...
‣ lgt8f
◦ abi.cpp
◦ CDC.cpp
◦ HardwareSerial.cpp
◦ HardwareSerial0.cpp
◦ HardwareSerial1.cpp
◦ HardwareSerial2.cpp
◦ HardwareSerial3.cpp
◦ hooks.c
◦ IPAddress.cpp
◦ main.cpp

• Error at line 51 in file main.cpp:
‣ 'PMX1' was not declared in this scope

• Error at line 52 in file main.cpp:
‣ 'PMX0' was not declared in this scope

• Error at line 146 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 146 in file main.cpp:
‣ 'F2XEN' was not declared in this scope

• Error at line 148 in file main.cpp:
‣ 'TC2XS0' was not declared in this scope

• Error at line 149 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 149 in file main.cpp:
‣ 'TC2XS0' was not declared in this scope

• Error at line 152 in file main.cpp:
‣ 'F2XEN' was not declared in this scope

• Error at line 162 in file main.cpp:
‣ 'DTR1L' was not declared in this scope

• Error at line 163 in file main.cpp:
‣ 'DTR1H' was not declared in this scope

• Error at line 166 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 166 in file main.cpp:
‣ 'F2XEN' was not declared in this scope

• Error at line 168 in file main.cpp:
‣ 'TC2XS1' was not declared in this scope

• Error at line 169 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 169 in file main.cpp:
‣ 'TC2XS1' was not declared in this scope

• Error at line 172 in file main.cpp:
‣ 'F2XEN' was not declared in this scope

• Error at line 175 in file main.cpp:
‣ 'TCCR3B' was not declared in this scope

• Error at line 177 in file main.cpp:
‣ 'TCCR3C' was not declared in this scope

• Error at line 177 in file main.cpp:
‣ 'DTEN3' was not declared in this scope

• Error at line 178 in file main.cpp:
‣ 'TCCR3A' was not declared in this scope

• Error at line 178 in file main.cpp:
‣ 'COM3B0' was not declared in this scope

• Error at line 180 in file main.cpp:
‣ 'TCCR3C' was not declared in this scope

• Error at line 180 in file main.cpp:
‣ 'DTEN3' was not declared in this scope

• Error at line 181 in file main.cpp:
‣ 'TCCR3A' was not declared in this scope

• Error at line 181 in file main.cpp:
‣ 'COM3B0' was not declared in this scope

• Error at line 182 in file main.cpp:
‣ 'DTR3A' was not declared in this scope

• Error at line 183 in file main.cpp:
‣ 'DTR3B' was not declared in this scope

• Error at line 224 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 224 in file main.cpp:
‣ 'TC2XF1' was not declared in this scope

• Error at line 226 in file main.cpp:
‣ 'ICR3L' was not declared in this scope

• Error at line 227 in file main.cpp:
‣ 'TCCR3B' was not declared in this scope

• Error at line 276 in file main.cpp:
‣ 'TCKCSR' was not declared in this scope

• Error at line 276 in file main.cpp:
‣ 'TC2XF1' was not declared in this scope

• Error at line 278 in file main.cpp:
‣ 'TCCR3B' was not declared in this scope

Compiling Failed

majenkotech, 2019-09-01:
That does look familiar. I think I saw the same thing at one point, and the cause was it including the wrong variant files in the package. I thought I'd fixed it already. Maybe it was for one of the other boards - though I'm sure I had all the boards compiling at least a basic empty sketch.

Kashouryo, 2020-02-25:
Here is an update.
GitHub user dbuezas made a package manager version of Larduino HSP.
Here's the URL: https://github.com/dbuezas/lgt8fx

Hopefully, this repo will make the porting process way much easier.

navr32, 2021-06-09:
Hello very good job ! 
Could you  add the  Dbuezas  arduino addon because the hsp is too buggy...and don't work...many many thanks
Have a nice days.

majenkotech, 2021-06-09:
I'd rather not. There are too many cores now, and keeping up to date with them all is impossible for me. I am working on a method of using `arduino-cli` as a compilation back-end for working with cores that aren't directly included with UECIDE.

navr32, 2021-06-09:
Ok it is a very good idea.
The Lgt328 is effectively working with the arduino-cli and arduino-ide 2.0 beta...

It is possible to help you ? Or it 'is possible to update myself the uecide  with the dbuezas package ? 

I have look at but i have found that all the support file all cores is not available directly on git but that uecide download
them directly from it's own servers so difficult to see what to do to add a core.

Have a nice day thanks for fast reply and congratulation for the very good jobs.

----- Mail original -----

majenkotech, 2021-06-09:
Building a core is a long and involved process (which is why I don't want to do it any more). 

The existing lgt8f build system is here: https://github.com/UECIDE-Cores/lgt8f

It's based around the Debian build system (dpkg-buildpackage) which uses `debian/rules` to tell it what to do. There's other scripts in there too to generate the board definitions. The UECIDE magic is in the `debain/core.txt` file which tells UECIDE what to do (likewise the board.txt file in each board folder).

The Arduino AVR core, which uses a package JSON file to find the files to download, is here: https://github.com/UECIDE-Cores/arduino-avr - that relies on a custom tool I wrote (but haven't published) and need to replace with `jq`.

Like I said... complex and long winded, and I hate it ;)

navr32, 2021-06-10:
Many thanks again . it is very annoying that's so complex...and long..and that you have to build custom tools..and
without this tool it is again very very hard to do it..
So it is better i drop it  i think..

----- Mail original -----